motioneye-project/motioneyeos #1521: Email Notification sever settings are lost when Email notifications are turned off!

wb666greene, 2018-05-24:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
Yes

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.
Yes

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).
20171003 upgraded over the air two days ago which is 201804xx I believe

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).
Raspberry PiZero-W
### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).
V4L2 PiCamera Module

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).
WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
None

### Log Files

I consider the following log files relevant to this issue:
None
 
This is an operational issue.

I set up Email Motion Notifications, the test worked and I received Email for motion events with attached photos while it was running in this mode.  I turned Motion Notifications off.  All the Email settings were lost when I turned Email notifications back on again.  I hope this is not intentional.

ccrisan, 2018-05-28:
This is unfortunately by design. More precisely, the presence of email settings in the webhooks list indicates that email notifications are enabled. Consequently, their absence indicates disabo email notifications.

wb666greene, 2018-06-15:
Thanks for the answer.
Then I guess this issue should be closed.

closed 2018-06-15 by wb666greene