motioneye-project/motioneyeos #1522: Wifi setup is much more difficult that it has to be!

wb666greene, 2018-05-24:
This is an operational issue with MotioneyeOS 20180224, probably earlier versions as well, but I got "lucky" and my wifi wpa_supplicant.conf worked the first time I installed MotioneyeOS on a PiZeroW using 2017xxxyy

Problem is if the wifi doesn't connect you are in a boot loop and can't do anything.

Fixing wifi connection issues would be made a lot easier if MotioneyeOS tried to connect to the wired Ethernet first and then fall back to the WiFi if the wired connection is not there -- the same behavior as Raspbian.

I'm currently using the Pi2 version on a Pi2.  I've tried two wifi dongles that are know to work on Raspbian -- a Patriot 11N and an "official" Pi WiFi dongle.

I set the WiFi SSID and PSK using the Web GUI using the wired Ethernet on the Pi2 after confguring Motioneye OS and was in a boot loop upon rebooting :(

Pulling the SD card on a Pi2 or better is generally not a big deal but on a PiZeroW with a Pi Camera module and the "Official Pi Zero case" (which is a nice compact setup and easy to mount with double sided tape or whatever) its quite the hassle to remove the SD card.

I just made a dumb typo :(    so it was easy to fix once I pulled the SD card.   But this can be a big obstacle  for someone running Windows who can't mount the ext4 partition containing /etc/wpa_supplicant.conf that ends up as /data/etc/wpa_supplicant.conf if you manage to ssh in, which would be easy to do if the wired connected was tried first to prevent the bootloop.

jasaw, 2018-05-25:
I agree with you that the wifi setup user experience needs to be improved. I proposed setting up a wifi access point when `/etc/wpa_supplicant.conf` is not found, see https://github.com/ccrisan/motioneyeos/issues/1422. It won't solve the reboot cycle caused by wrong wifi password (unless we can somehow detect this). Connecting via Ethernet and use wifi as fallback as you have suggested will help some users, but Pi Zero W users without Ethernet dongle will still get reboot cycle.

When issue https://github.com/ccrisan/motioneyeos/issues/1421 is fixed, there won't be a need to access '/etc/wpa_supplicant.conf' directly.

darkplatinum, 2018-06-07:
Basically what I have done is got the motion eye OS on top of raspbian. You can easily setup wifi with the raspbian GUI, then later if you need to use less memory you boot to CLI (command line interface). Works like a treat.

alphanumeric007, 2018-08-07:
Why does it endlessly reboot when no network is found, in the first place? That is so frustrating and I just don't see why its like that? I had a monitor keyboard and mouse connected on first boot and all I got was endless reboots? Why can't you just go in after it boots up and setup your WIFI? IMHO having to edit and add a wpa_supplicant file is not the answer. Its way to easy to have an editing error, especially if your doing it on a Windows PC. 
At the very least have it prompt for your wifi credentials on boot up. I ended up plugging in a USB hub with an ethernet connection so I could get in and setup a wifi connection via the motioneye setup menu. That's is not ideal. IMHO the user experience needs to be improved.

metalshreds, 2018-08-24:
Just thought I'd chime in with my 2 cents. setup wifi through ethernet, and accidentally added a blank space at the beginning of the password, constant reboots and "wifi: disconnected" prompt. and no way to get back and fix the password, even plugging in through ethernet didn't work. Luckily I had a linux machine to edit the wpa_supplicant file. Love the project and happy to contribute with feedback :)

codesterribly, 2018-09-28:
Same feelings. This behavior needs to be changed or a flag set 

'bootloopforever=1' if wifi is not found.

I see the reasoning behind it, but honestly, it makes it a real pain to fix since the camera was working fine for a few hours, so I put it in the camera housing. Now, it's bootlooping, and it's a real pain to get the msd card out or to attach usb/hdmi to it.

ccrisan, 2018-09-29:
The next release will have a flag to *enable* this boot loop, which will be disabled by default.

ricso, 2019-05-28:
> Basically what I have done is got the motion eye OS on top of raspbian. You can easily setup wifi with the raspbian GUI, then later if you need to use less memory you boot to CLI (command line interface). Works like a treat.

How do you install Motion Eye OS on top of Raspbian and how do you launch the Raspbian GUI to connect to Wifi manually?  I am trying to connect my PiZero with Motion Eye OS to a hotel wifi, which needs to be done manually, since I need to enter my room number and surname before I can connect

alphanumeric007, 2019-05-28:
This may be what your looking for, [  https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian   ](url)
Didn't work for me last time i tried it. 
ccrisan can confirm but I do believe the Motion Eye main download is based on Raspbian Lite. No GUI installed? If not, you can run sudo raspi-config from the command line and switch it so it boots to the Desktop GUI. Once there manually setup your WIFI, then switch it back to booting to command line.