motioneye-project/motioneyeos #1523: motioneye does not start on startup (raspbian)

durahsel, 2018-05-27:
Hi,
I've tried several methods (crontab, autostart with .desktop script ....) but none of them works for me, after boot I've each time to manually start motion eye with this script (which is 'chmod +x')

System

motionEye Version | 0.39
Motion Version | 4.1.1
OS Version | Raspbian 9.4

/home/pi/Desktop/Scripts/motionEyeinitSansSudo.sh

```
#!/bin/bash
cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
systemctl daemon-reload
systemctl enable motioneye
systemctl start motioneye
```

here is my last try with /etc/rc.local method
/etc/rc.local file is 

```
#!/bin/bash
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.
bash /home/pi/Desktop/Scripts/motionEyeinitSansSudo.sh
exit 0
```

also try this one 

```
#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
# In order to enable or disable this script just change the execution
# bits.
# By default this script does nothing.
sh /home/pi/Desktop/Scripts/motionEyeinitSansSudo.sh
exit 0
```

I know that the script is starting after reboot (trying to add some log file creation at the end of the motionEyeinitSansSudo.sh script and it works) but when I browser motioneye, It does not work. If  I manually start the script after reboot, then it's works

Any idea to make the motionEye works on startup ?

thanks !

jasaw, 2018-06-06:
This is a motionEyeOS repo. Please ask this question in motionEye repo: https://github.com/ccrisan/motioneye/issues

durahsel, 2018-06-06:
oups ... thanks (transfered to the good forum [here](https://github.com/ccrisan/motioneye/issues/900)

closed 2018-06-06 by durahsel