motioneye-project/motioneyeos #1524: Setting up MotionEye OS on Pi 3 B with a Kuman OV5647 camera works initially but as soon as I change something it stops working.

dax2112, 2018-05-27:
So I installed the below items and all works fine.  As soon as I change the admin password it will not recognize the new or old password, I have to re-flash the disk and start over.

Any help would be appreciated!

When I added the IR lights to the camera, MotionEye says there is no camera connected.

I am running motionEyeOS version: motioneyeos-raspberrypi3-20180401

I am using the following board/model: Pi 3 B

I am using the following type of camera: V4L2  ??  

My camera model is: kuman for Raspberry PI Camera Module 5MP 1080p OV5647 Sensor HD Video Webcam Supports Night Vision SC15

My motionEyeOS unit is connected to the network via: WiFi but I can plugin Ethernet

dax2112, 2018-05-30:
Beuller, beuller??  Is my issue not worthy of a response?

jasaw, 2018-05-30:
@dax2112 Looks like you have 2 issues:
1. Changing admin password locks you out of the system. Is it the same as issue https://github.com/ccrisan/motioneyeos/issues/1317? Can you please provide the step-by-step to reproduce the problem? 
2. IR lights causing camera detection to fail. I would start by checking the CSI cable connection, then check the Pi power supply (those IR lights take a fair bit of power), check your power supply microUSB connector. Try with only one IR light attached.

dax2112, 2018-06-02:
So Yes it's exactly like issue 1317, but there was not a resolution to that case?

Can I install motionEye software on top or in NOOBS?

jasaw, 2018-06-04:
@dax2112 Looks like the admin password problem has just been fixed, will be included in the next release.

Yes you can install motionEye on NOOBS. Follow the [motionEye installation guide](https://github.com/ccrisan/motioneye/wiki/Installation).