motioneye-project/motioneyeos #1525: wpa_supplicant.conf reset after changing settings

fei0316, 2018-05-28:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20180401

### Board Model

I am using the following board/model: Raspberry Pi 3

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

I am using WPA2 EAP (Enterprise) for my WiFi, and I can connect successfully using the following configuration:

(irrelevant parts ommitted)

```
network={
key_mgmt=WPA-EAP
ssid="xxxxx"
pairwise=CCMP
group=CCMP
auth_alg=OPEN
eap=PEAP
proto=RSN
identity="xxxx"
password="xxxxxx"
}
```

However, after changing settings from the web ui (that are unrelated to WiFi, I didn't touch the settings for WiFi), the 
`key_mgmt=WPA-EAP `
changes to 
`key_mgmt=NONE`
and thus making the connection fail on boot.

ccrisan, 2018-05-29:
Unfortunately motionEye doesn't play well with complex WiFi setups. It has a very basic wpa_supplicant.conf reading/writing procedure that will do that with your file. The solution here would be to recognize other key_mgmt options and to leave them as they are.

fei0316, 2018-05-30:
Sure, this solution would be good enough since EAP users should really know how to make the config file and support for EAP in the UI is really not that important. I would say if an unexpected key_mgmt option is found in wpa_supplicant.conf, the WiFi option could be greyed out and left untouched when saving other settings.

Thank you very much and looking forward to the update!

georgia-lawlor, 2023-02-03:
Hi, has any progress been made on adding support for more complex WiFi settings? Anyone on a college campus would struggle with this. I am a graduate students and trying to set up 24/7 monitoring cameras but cannot connect to the college wifi. The monitoring room in question has no ethernet options so I had to use an additional Pi with Raspbian to forward the WiFi to a router in order for the cameras to have internet connection. This could all be solved is support were added for more complex wifi connections. Specifically anything similar to "eduroam". 

My wpa_supplicant.conf file looks like this:
```
country=US
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
    ssid="XXXXX"
    scan_ssid=1
    key_mgmt=WPA-EAP
    group=CCMP TKIP
    eap=PEAP
    identity="XXXXX@XXX.edu"
    password="XXXXX"
    phase1="peapver=0"
    phase2="MSCHAPV2"
}
```

starbasessd, 2023-02-03:
I believe it will depend on what base they choose to use for the next generation. Currently (through dev20201026) is/was built on a product called BuildRoot, supplemented with ThingOS. 
I have seen no attempted development by the new devs for motionEyeOS and the Python3.x update. 
I do not know if the wpa_supplicant from the latest BuildRoot can support the extra functionality your setup needs. I would seriously consider shifting to something like a stripped out RaspberryPiOS lite, or Debian lite on the Pi, and run motionEye from [here](https://github.com/motioneye-project/motioneye/wiki/Installation)