motioneye-project/motioneyeos #1527: Update is not applying

patrickhener, 2018-05-29:
### motionEyeOS Version

20180101

### Board Model

Raspi 3B

### Camera

Raspi Cam NoIR

### Network Connection

Ethernet Cable

### Peripherals

None

### Problem

When checking for updates there is the new update 20180401 available. Updater then runs and finally tells the update was successful. But when checking the version it is still 20180101 even after rebooting.

New features are not applied as well (for example: In Browser Video playback).

patrickhener, 2018-05-29:
### motioneye.log

2018-05-29 15:05:43: [motioneye]     INFO: updating to version 20180401...
2018-05-29 15:05:45: [motioneye]     INFO: stopping motioneye watch script
2018-05-29 15:05:45: [motioneye]     INFO: stopping netwatch script
2018-05-29 15:05:45: [motioneye]     INFO: downloading firmware version 20180401
2018-05-29 15:05:56: [motioneye]     INFO: extracting firmware
2018-05-29 15:05:56: [motioneye]     INFO: flashing boot partition
2018-05-29 15:05:56: [motioneye]    ERROR: Exception in callback <functools.partial object at 0x74161480>
Traceback (most recent call last):
  File "usr/lib/python2.7/site-packages/tornado/ioloop.py", line 605, in _run_callback
  File "usr/lib/python2.7/site-packages/tornado/stack_context.py", line 277, in null_wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/platformupdate.py", line 58, in perform_update
Exception: firmware flash boot failed

ccrisan, 2018-05-29:
@patrickhener you must manually upgrade to an intermediate version (20180224), given that things have changed a bit in the upgrade process. You can do that by typing `fwupdate upgrade 20180224` in the command line. If that works, you can do the same for `20180401`. Or you can use the UI then.

patrickhener, 2018-05-30:
Works like a charm. Thanks.

closed 2018-05-30 by ccrisan

darren-harrison, 2021-02-09:
Awesome, that helped me get off a very old version as well. Thanks.