motioneye-project/motioneyeos #1528: anyway to turn off ethernet LEDs on Pi 2 with Motioneyeos?

shadowfxd, 2018-05-30:
Can someone help me or point me in the right direction to turn off the ethernet LEDs?

shadowfxd, 2018-06-01:
it did some weird flashing but didn't stick. The LEDs are still on after fully booted.

jasaw, 2018-06-01:
According to this [forum](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=72070&start=25),  `eth_led0` and `eth_led1` are supported on RPi 3B+ only at the moment. I don't know if this feature will make its way to the other RPi models.

In the mean time, you still can disable Eth LEDs but it's more complicated. You probably need to set up your MotionEyeOS cross-compilation environment, compile a little program, and copy it to the Pi. Alternatively, you could run Raspbian + MotionEye, and compile this program locally on the Pi.

Refer to https://github.com/raspberrypi/linux/issues/2280 for more info.

shadowfxd, 2018-06-13:
I tried it on a Pi3B+ and it turned off the LEDs. Thank you! Hopefully it somehow gets added to older Pi's.

jasaw, 2018-06-14:
@ccrisan Should we add `lan951x-led-ctl` package to allow turning off Ethernet LEDs on older RPi?

@shadowfxd Thank you for testing and reporting back. I've updated [our turn off LEDs wiki](https://github.com/ccrisan/motioneyeos/wiki/Turn-Off-LEDs).

shadowfxd, 2018-06-14:
@ccrisan please consider what @jasaw is asking since the less LEDs the better for some of us trying to hide these cameras in windows.

ccrisan, 2018-06-20:
@jasaw I don't see why not. As far as I can tell, it's a fairly straight Makefile package that should compile without too much struggle on a RPi setup.

ccrisan, 2018-06-24:
https://github.com/ccrisan/thingos/commit/b875763a6ab0bbf0066856e41869b130249329f2 and https://github.com/ccrisan/thingos/commit/bd02abf38d536094ba4a8f43b3dded7197af45a1 add support for `lan951x-led-ctl` for all Raspberry PI boards.

The code will be merged into motionEyeOS after the next *stable* release.

shadowfxd, 2018-06-24:
thank you so much @ccrisan !

shadowfxd, 2018-06-27:
I forgot to ask, any ETA on the next stable version @ccrisan?

shadowfxd, 2018-07-04:
I just tried the code on the newest stable release (20180627) and the ethernet lights didn't turn off.

ccrisan, 2018-07-04:
@shadowfxd as I said before, the change will be merged in *after this stable release*. 20170627 is a release that we've been working on for some time to reach a stable state.

You'll soon be able to test our next nightly build which should contain the `lan951x-led-ctl` package. However, don't expect those leds to automatically turn off. You'll need to manually call the command with appropriate arguments and, if it works as expected, you can add it to `/data/etc/userinit.sh`.

shadowfxd, 2018-07-04:
perfect, thank you!

ccrisan, 2018-07-12:
It should work but I haven't tested it yet. What command did you try? What error did you get?

shadowfxd, 2018-07-12:
I did exactly as @jasaw showed in a earlier post that I got working on Pi 3.

jasaw, 2018-07-13:
@shadowfxd Which method did you use? The `dtparam` method only works on Pi 3B+. On other models of RPi, you need to run something like this `lan951x-led-ctl --fdx=0 --lnk=0 --spd=0`, but I haven't tested it.

shadowfxd, 2018-07-13:
I have no knowledge of this method. Can you please help me learn it?

jasaw, 2018-07-13:
@shadowfxd Try running `lan951x-led-ctl --fdx=0 --lnk=0 --spd=0` command from your SSH console. If it works, add it to `/data/etc/userinit.sh`

shadowfxd, 2018-07-13:
I tried it in putty, it worked. Added it to `/data/etc/userinit.sh` and reboot. It worked! Thank you @jasaw !

jasaw, 2018-07-13:
@shadowfxd @ccrisan I've updated the [Turn off LEDs wiki](https://github.com/ccrisan/motioneyeos/wiki/Turn-Off-LEDs).

ccrisan, 2018-07-13:
Great then. Can we close the issue now?

shadowfxd, 2018-07-13:
yes, thank you @ccrisan and @jasaw

MartoPW, 2018-11-01:
# Disable the ACT LED.
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off

This is not working on Pi 3, can someone advice what might be the problem (the other two commands from the article successfully disabled the power LED but the camera LED is still ON)
Thank you

shadowfxd, 2018-12-10:
when will you add it back in?

shadowfxd, 2018-12-11:
thank you @ccrisan and @jasaw as usual. Closing again, since issue is resolved.

closed 2018-12-11 by shadowfxd

seamusdemora, 2022-03-16:
@ccrisan :
> @shadowfxd as I said before, the change will be merged in _after this stable release_. 20170627 is a release that we've been working on for some time to reach a stable state.
>
> You'll soon be able to [...]

I've got a model `Raspberry Pi Model B Plus Rev 1.2` currently running RPi OS `buster`.  I want to run it 'dark'/no LEDs. I've turned off the `pwr` & `act` LEDs using `dtparam` in `/boot/config.txt`. I've been searching for a solution to douse the two ethernet LEDs, which led me here. When I saw this post & reference to a `lan951x-led-ctl package`, I assumed this was something I could install using `apt`, but when I tried, I got `E: Unable to locate package lan951x-led-ctl`. 

Can someone briefly explain the situation with `lan951x-led-ctl` ? 
Or point me to another method/resource for turning those LEDs off?

Thnx,
~S