motioneye-project/motioneyeos #1530: system won't start up after power is turned off

adrian0, 2018-06-03:
motionEyeOS version: 20180401 on Raspberry Pi Zero W with MMAL Camera
Connecting over WiFi
Unable to collect log files

While in normal operation (No other issues of defects apparent), after power is cycled (wall switch is turned off and back on), motioneyeos does not restart.
Cannot even SSH to it.

jasaw, 2018-06-04:
This looks like a duplicate of https://github.com/ccrisan/motioneye/issues/893.

alexandru-2016, 2018-06-24:
exact thing happened to me. I interrupted the power supply to my PI zero and now the system doesn't boot up. I attached a screen but there is no signal on hdmi, and the device doesn't connect to the network.

ccrisan, 2018-06-24:
@adrian0, as @jasaw pointed out, this is a duplicate of https://github.com/ccrisan/motioneye/issues/893. I believe you realized that this is rather a motionEyeOS issue than a motionEye one and you reopened it here.

However there are a lot less details this time and I would expect you to answer my questions asked in https://github.com/ccrisan/motioneye/issues/893.

ccrisan, 2018-06-24:
@alexandru-2016 please let us know your answers to my questions in https://github.com/ccrisan/motioneye/issues/893.

jasaw, 2018-06-25:
If the HDMI doesn't show anything, that means the Pi can't even read the boot partition, but motionEyeOS doesn't write to boot partition during normal operation.
I wonder if the entire SD card is corrupted. If a power loss occurs during a write operation within the SD card, there's always a chance to corrupt the entire SD card. I would plug the SD card to a computer and run `fsck`.

alexandru-2016, 2018-06-25:
I could read the sd card from my pc, because I backed up the wifi settings
file.

ccrisan, 2018-06-25:
Guys, the Raspberry PI will not display anything on HDMI unless powered up with the monitor connected. Connecting the monitor after that and not seeing anything on it may be inconclusive.

alexandru-2016, 2018-06-25:
I did try to bootup the pi a few times after plugging in the hdmi, but it
didn't display anything. Only sign of activity was the board leds
flickering.

ccrisan, 2018-06-25:
@alexandru-2016 that can mean 3 things, in my opinion:

 * the SD card got corrupted somehow or physically damaged
 * your PI has a hardware fault
 * your monitor has a hardware fault

You could try using another board if you have one at your disposal. Or you could try recovering the entire SD card contents (using `dd`) from your PC, archive it and send it to me somehow so I can take a look at it.

Instagit, 2018-07-10:
I can confirm this on the Pi Zero W and it's not the hardware. It's fully reproducible and happens every time the power is cut or the system fails to boot for any reason (full disk for example).

After a power failure, the system tries to load a wired network config instead of wireless. Reflashing fixes the issue, but there must be another way.

Last successful boot:

````log
---- booting motionEyeOS 20171008 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting eudev: done
 * Starting watchdog: done
 * Starting wpa_supplicant: done (brcmfmac)
 * Configuring wireless network: 192.168.178.8/24
 * Setting static gateway to 192.168.178.1: done
 * Setting static DNS server to 8.8.8.8: done
 * Setting current date using http: Mon Feb 12 16:09:05 CET 2018
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface wlan0 has IP address 192.168.178.8/24
 # Default gateway is 192.168.178.1
 # DNS server address is 8.8.8.8
````
After power cut:

````
---- booting motionEyeOS 20171008 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting eudev: done
 * Starting watchdog: done
 * Configuring wired network: no device
````

Let me know if you need further information.

jasaw, 2018-07-11:
@Instagit Please copy the entire corrupted SD card to your PC, using tools like `dd`, and somehow share it with us.

Instagit, 2018-07-11:
Ok. Where's the wireless info stored? I want to redact that.

Instagit, 2018-07-11:
I sent a mail with the download link for the SD card contents to @ccrisan and @jasaw. Thanks for looking into this.

ccrisan, 2018-07-11:
@Instagit your data/wpa_supplicant.conf file has somehow become empty. If you say your disk was full, then it makes sense, since motionEye tries to overwrite it (and other config files) upon applying preferences. And with a full disk, it simply wouldn't be able to actually write stuff to it.

jasaw, 2018-07-12:
@ccrisan motionEye should always write the new config file to a temp file in the same partition, then rename (`mv` operation) the temp file to the actual config file. This way, a failed write will not corrupt the config file.

ccrisan, 2018-07-12:
@jasaw motionEye fucks up wpa_supplicant.conf in many ways, actually :) But yes, a first step would be to write to a temporary file to overcome situations like this one.

Instagit, 2018-07-22:
@ccrisan @jasaw Fixing the wpa_supplicant.conf solved the issue. Thanks!

lancstanley, 2018-08-05:
Hi Ccrisan

Had the same issue with three rpi zero w's..  all work fine, after setting them up for the first time (adding wpa_supplicant.conf to the boot partition after imaging) and I will admit to yanking the power lead out _(however daft it seems)_ because I wanted to know how to fix it.. in the real world, any number of things might cause a power outage.. for cctv, you do want it to restart unattended..

Installed Rasbian then manually installed Motioneye, following [the github instructions](https://github.com/ccrisan/motioneye/wiki/installation).. same issue again on power cycling.. unable to see the pi's reserved ip address on LAN..

I've done two things that seem to have remedied it.. 

(connecting pi zero to a display) and changing the network settings, ticking the box for wait for network on boot...

then:
        sudo nano /etc/network/interfaces 

[code]
auto wlan0
iface wlan0 inet dhcp 
                wpa-ssid {ssid}
                wpa-psk  {password}
[/code]

folllowed by:

[code]sudo dhclient wlan0[/code]

(found this idea @ post 69: https://unix.stackexchange.com/questions/92799/connecting-to-wifi-network-through-command-line)

..just yanked the power again, no probs at all on reboot.. if you'd like any Log files etc..
 
Thanks for a great rpi cctv solution! (I will donate when I can) It's going to feed an iframe video, displayed on a local web site/server.. ajax.. php and robotics.. finally getting in touching distance of having my 3D printed sentry/rover for the winter... armed with flood lights and verious sensors etc etc... Motioneye saves me a huge amount of time.. very nice/clean interface too. 

Best Regards, L.

jasaw, 2018-08-06:
@lancstanley Everytime you pull the power from a motionEyeOS device, you run the risk of corrupting the entire SD card. There are ways to minimize that, or to avoid that entirely, like minimizing writes to SD card (log to RAM disk, and so on...), and write media to a network drive to avoid writing to SD card.
With Raspbian, the risk of SD card / filesystem corruption is even higher, unless you made your root filesystem read-only.

I agree this is a serious issue for a CCTV solution, but it requires a multi-layered solution to beef it up.
1. MotionEye has to use a "write & move" method when updating config files. See my earlier post. I believe this shouldn't be difficult to implement.
2. Possibly give an option on the web UI to log to RAM. This minimizes writes to SD card.
3. Put all config files in a separate partition that is mounted as synchronous writable.

lancstanley, 2018-08-06:
Thanks for the explanation/guidance Jasaw, much appreciated.  Definitely will look into the above.. thinking about a micro controller with an i2c voltage sensor (have a couple doing nothing) also wondered if it was worth trickle charging 8 AA's using 12v and powering the pi zero's with DC to DC buck converters.  That way, at least you'd get an email saying that power supply had changed from mains to battery and or an hour or two to address any problems.  

I do like Motioneye on rpi, seems more than enough for my needs, even on the pi zero, which continues to amaze me.

jasaw, 2018-08-07:
@lancstanley You might want to make sure your AA batteries + DC-DC combo has good enough transient response to power a Raspberry Pi. For battery backup, I would use a lithium ion battery, probably a 18650 cell, DC-DC boost, and a battery charger module.

coderpop, 2019-04-28:
This just happened to me with wired network.

Brief power interruption due to storm and now none of the 5 motioneyeos cameras are operable. All other pi devices recovered and working normally, so issue with meos. Cameras are mix of pi 3, pi zero, pi 3+. Is there no fix availble??

ccrisan, 2019-04-28:
That doesn't normally happen. Please provide some logs and let's see what actually has happened, so we can find a solution.