motioneye-project/motioneyeos #1531: Building motioneyeos from source fails as of 03-06-2018 (Probably a make issue)

Byrkoet, 2018-06-03:
Building motioneyeos from source fails as of _03-06-2018_, even though I made sure to have all prerequisites for buildroot and followed the guide on the wiki. This seems to be a make issue, perhaps I haven't configured it properly. Anyone who can help?

### motionEyeOS Version

I am running motionEyeOS version: 20180602.

### Board Model

I am using the following board/model: Raspberry PI 3B.

### Camera

I am using the following type of camera: V4L2

My camera model is: Kuman SC15-Webcams-UK.

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Log Files

This is the bottom part of the output of `./build.sh raspberrypi`:

(The complete output may be found here https://gist.github.com/Byrkoet/db9d68084980d8661edc5c029d1d299b)

```
unix.c: In function ‘main’:
unix.c:312:47: warning: ‘%s’ directive output may be truncated writing up to 255 bytes into a region of size 58 [-Wformat-truncation=]
    snprintf(fname, 80, "/proc/acpi/ac_adapter/%s/state",
                                               ^~
In file included from /usr/include/stdio.h:862:0,
                 from unix.c:15:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:64:10: note: ‘__builtin___snprintf_chk’ output between 29 and 284 bytes into a destination of size 80
   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        __bos (__s), __fmt, __va_arg_pack ());
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
	CC badblocks.c
	CC util.c
	CC dirinfo.c
	CC dx_dirinfo.c
	CC ehandler.c
	CC problem.c
	CC message.c
	CC quota.c
	CC recovery.c
	CC region.c
	CC revoke.c
	CC ea_refcount.c
	CC rehash.c
	CC logfile.c
	CC sigcatcher.c
	CC readahead.c
	CC extents.c
	SUBST e2fsck.8
	SUBST e2fsck.conf.5
	LD e2fsck
making all in debugfs
	MK_CMDS debug_cmds.c
	CC debugfs.c
	CC util.c
	CC ncheck.c
	CC icheck.c
	CC ls.c
	CC lsdel.c
	CC dump.c
	CC set_fields.c
ls.c: In function ‘print_filename’:
ls.c:60:8: warning: ‘<encrypted (’ directive output truncated writing 12 bytes into a region of size 1 [-Wformat-truncation=]
      "<encrypted (%d)>", len);
       ~^~~~~~~~~~~
ls.c:60:6: note: directive argument in the range [0, 255]
      "<encrypted (%d)>", len);
      ^~~~~~~~~~~~~~~~~~
In file included from /usr/include/stdio.h:862:0,
                 from ls.c:9:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:64:10: note: ‘__builtin___snprintf_chk’ output between 16 and 18 bytes into a destination of size 1
   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        __bos (__s), __fmt, __va_arg_pack ());
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
	CC logdump.c
	CC htree.c
	CC unused.c
	CC e2freefrag.o
	CC filefrag.c
	MK_CMDS extent_cmds.c
	CC extent_inode.c
	CC zap.c
	CC create_inode.o
	CC quota.c
	CC xattrs.c
./../misc/create_inode.c: In function ‘set_inode_xattr’:
./../misc/create_inode.c:140:9: warning: implicit declaration of function ‘llistxattr’; did you mean ‘lstat64’? [-Wimplicit-function-declaration]
  size = llistxattr(filename, NULL, 0);
         ^~~~~~~~~~
         lstat64
./../misc/create_inode.c:176:16: warning: implicit declaration of function ‘lgetxattr’; did you mean ‘getdate_r’? [-Wimplicit-function-declaration]
   value_size = lgetxattr(filename, name, NULL, 0);
                ^~~~~~~~~
                getdate_r
./../misc/create_inode.c: At top level:
./../misc/create_inode.c:399:18: error: conflicting types for ‘copy_file_range’
 static errcode_t copy_file_range(ext2_filsys fs, int fd, ext2_file_t e2_file,
                  ^~~~~~~~~~~~~~~
In file included from ./../misc/create_inode.c:19:0:
/usr/include/unistd.h:1110:9: note: previous declaration of ‘copy_file_range’ was here
 ssize_t copy_file_range (int __infd, __off64_t *__pinoff,
         ^~~~~~~~~~~~~~~
Makefile:422: recipe for target 'create_inode.o' failed
make[4]: *** [create_inode.o] Error 1
make[4]: *** Waiting for unfinished jobs....
Makefile:404: recipe for target 'all-progs-recursive' failed
make[3]: *** [all-progs-recursive] Error 1
Makefile:339: recipe for target 'all' failed
make[2]: *** [all] Error 2
package/pkg-generic.mk:234: recipe for target '/home/test/motioneyeos/output/raspberrypi3/build/host-e2fsprogs-1.43.7/.stamp_built' failed
make[1]: *** [/home/test/motioneyeos/output/raspberrypi3/build/host-e2fsprogs-1.43.7/.stamp_built] Error 2
Makefile:79: recipe for target '_all' failed
make: *** [_all] Error 2
```

jasaw, 2018-06-06:
Try upgrading e2fsprogs to version 1.44.1.

Byrkoet, 2018-06-06:
It says I have latest:
```
test@test-VirtualBox:~$ sudo apt install e2fsprogs
Reading package lists... Done
Building dependency tree       
Reading state information... Done
e2fsprogs is already the newest version (1.44.1-1).
```
Could it have something to do with the config?

```
make[4]: Nothing to be done for 'all'.
/usr/bin/make progs
make[4]: 'util/subst.conf' is up to date.
make[4]: 'lib/config.h' is up to date.
make[4]: 'lib/dirpaths.h' is up to date.
make[4]: 'lib/ext2fs/ext2_types.h' is up to date.
make[4]: 'lib/blkid/blkid_types.h' is up to date.
make[4]: 'lib/uuid/uuid_types.h' is up to date.
make[4]: 'compile_et' is up to date.
make[4]: 'ext2_err.h' is up to date.
make[4]: 'prof_err.h' is up to date.
making all in lib/et
make[4]: Nothing to be done for 'all'.
making all in lib/ss
make[4]: Nothing to be done for 'all'.
making all in lib/e2p
make[4]: Nothing to be done for 'all'.
making all in lib/support
make[4]: Nothing to be done for 'all'.
making all in lib/ext2fs
make[4]: Nothing to be done for 'all'.
making all in intl
make[4]: Nothing to be done for 'all'.
making all in e2fsck
make[4]: Nothing to be done for 'all'.
making all in debugfs
	CC create_inode.o
	SUBST debugfs.8
./../misc/create_inode.c: In function ‘set_inode_xattr’:
./../misc/create_inode.c:140:9: warning: implicit declaration of function ‘llistxattr’; did you mean ‘lstat64’? [-Wimplicit-function-declaration]
  size = llistxattr(filename, NULL, 0);
         ^~~~~~~~~~
         lstat64
./../misc/create_inode.c:176:16: warning: implicit declaration of function ‘lgetxattr’; did you mean ‘getdate_r’? [-Wimplicit-function-declaration]
   value_size = lgetxattr(filename, name, NULL, 0);
                ^~~~~~~~~
                getdate_r
./../misc/create_inode.c: At top level:
./../misc/create_inode.c:399:18: error: conflicting types for ‘copy_file_range’
 static errcode_t copy_file_range(ext2_filsys fs, int fd, ext2_file_t e2_file,
                  ^~~~~~~~~~~~~~~
In file included from ./../misc/create_inode.c:19:0:
/usr/include/unistd.h:1110:9: note: previous declaration of ‘copy_file_range’ was here
 ssize_t copy_file_range (int __infd, __off64_t *__pinoff,
         ^~~~~~~~~~~~~~~
Makefile:422: recipe for target 'create_inode.o' failed
make[4]: *** [create_inode.o] Error 1
Makefile:404: recipe for target 'all-progs-recursive' failed
make[3]: *** [all-progs-recursive] Error 1
Makefile:339: recipe for target 'all' failed
make[2]: *** [all] Error 2
package/pkg-generic.mk:234: recipe for target '/home/test/motioneyeos/output/raspberrypi3/build/host-e2fsprogs-1.43.7/.stamp_built' failed
make[1]: *** [/home/test/motioneyeos/output/raspberrypi3/build/host-e2fsprogs-1.43.7/.stamp_built] Error 2
Makefile:79: recipe for target '_all' failed
make: *** [_all] Error 2
```

jasaw, 2018-06-07:
Sorry, I meant upgrading the e2fsprogs package in buildroot to version 1.44.1. e2fsprogs-1.43.7 has some compilation problem with certain versions of gcc.

Change this line to 1.44.1
https://github.com/ccrisan/motioneyeos/blob/7df75d1ad072ca9d967691b6a00e304118652826/package/e2fsprogs/e2fsprogs.mk#L7

If this solves your problem, we'll create a PR to fix this in master.

emartisoft, 2018-08-24:
In addition create new hash code for e2fsprogs-1.44.1 and write and edit it into e2fsprogs.hash file.
`` echo -n e2fsprogs-1.44.1.tar.xz | sha256sum``

e2fsprogs.hash file are same folder as package/e2fsprogs/e2fsprogs.mk
`` sha256 0ca164c1c87724df904c918b2d7051ef989b51de725db66c67514dbe6dd2b9ef  e2fsprogs-1.44.1.tar.xz``

jasaw, 2018-08-27:
@emartisoft Thanks for that. `e2fsprogs` version 1.44.1 fixes the compilation problem?

jasaw, 2018-12-01:
`e2fsprog` in `dev` branch has been upgraded to 1.44.2, so this should not be a problem anymore.

closed 2018-12-01 by jasaw

ccrisan, 2018-12-03:
As a side note, from now on, motionEyeOS's repo has a common git history with BuildRoot's. Therefore we can merge in their changes more easily.

gilsonmandalogo, 2019-06-07:
@jasaw This solution in `master` branch solves the build process. Shall I create a PR with this, or wait the merge of `dev` into `master`? Also, there are plans to update the `master` branch? It seems to be old.

ccrisan, 2019-06-07:
@gilsonmandalogo thanks for noticing that our master was left behind. I've now updated it to 20190427.