motioneye-project/motioneyeos #1532: Problem with Action Buttons

rockman55, 2018-06-03:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

ccrisan, 2018-06-09:
Your log says:

    2018-06-03 16:03:44: [motioneye]  WARNING: up_1: python: can't open file 'up1.py': [Errno 2] No such file or directory
    2018-06-03 16:28:40: [motioneye]  WARNING: down_1: command has finished with non-zero exit status: 1
    2018-06-03 16:28:40: [motioneye]  WARNING: down_1: Traceback (most recent call last):
    2018-06-03 16:28:40: [motioneye]  WARNING: down_1:   File "/data/etc/down_1", line 6, in <module>
    2018-06-03 16:28:40: [motioneye]  WARNING: down_1:     import Adafruit_PCA9685
    2018-06-03 16:28:40: [motioneye]  WARNING: down_1: ImportError: No module named Adafruit_PCA9685
    2018-06-03 16:28:42: [motioneye]  WARNING: down_1: command has finished with non-zero exit status: 1

You seem to require an extra python package, called `Adafruit_PCA9685`, which is not available on motionEyeOS. I'm afraid you'll have to move to Raspbian + motionEye if you really need to use it. Or, if you have the knowledge and time, add the package to a custom built motionEyeOS variant.

closed 2018-06-09 by ccrisan

rockman55, 2018-06-09:
Thanks, Calin for your response. I have built this package: https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian, and installed the Adafruit package. The action buttons will work with that installation, but it seems that many of the other functions of MotionEye are not there. Like check for update, backup and restore and others. Is there a way to do the Raspbian plus Motioneye and get all of the features that are in the original distribution?

Thanks for your help.

Danny aka rockman

ccrisan, 2018-06-09:
I'm afraid that that's not going to be possible.

rockman55, 2018-06-09:
What features would be available if I do a package of a custom built motionEyeOS variant? Are there instructions on how to do that? I have plenty of time for that.

Thanks 

Danny