motioneye-project/motioneyeos #1534: samba not starting

AlquistArj, 2018-06-06:
I'm unable to connect to the Samba file share. I tried using the latest version but no luck. See text from boot.log below. Any idea how to fix this?

---- booting motionEyeOS 20180602 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Starting wpa_supplicant: done
 * Configuring wireless network: dhcp
 * Setting current date using http: Wed Jun 6 13:57:52 BST 2018
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting smbd: failed
 * Starting nmbd: failed
 * Starting motioneye: done
 # Interface wlan0 has IP address 192.168.65.5/24
 # Default gateway is 192.168.65.254
 # DNS server address is 192.168.65.254

jasaw, 2018-06-06:
Please attach all the log files.

AlquistArj, 2018-06-06:
[dmesg (6).log](https://github.com/ccrisan/motioneyeos/files/2076551/dmesg.6.log)
[motioneye (5).log](https://github.com/ccrisan/motioneyeos/files/2076552/motioneye.5.log)
[boot (8).log](https://github.com/ccrisan/motioneyeos/files/2076553/boot.8.log)
[motion (3).log](https://github.com/ccrisan/motioneyeos/files/2076554/motion.3.log)

[messages (7).log](https://github.com/ccrisan/motioneyeos/files/2076559/messages.7.log)

jasaw, 2018-06-07:
Looks like your data partition is corrupted. Try running fsck on that partition from another machine.