motioneye-project/motioneyeos #1537: Google photos

darkplatinum, 2018-06-07:
Hi! Basically this is more of a request than an issue. 

So far I've started to setup my cctv system. Just waiting for my camera to arrive. During some tutorials I've seen your can send data to google drive. As google drive has free 15 GB of storage, it makes it a great choice for Free storage. But Google photos has UNLIMITED storage. Yes unlimited! Basically, in Google photo settings you change original quality to high quality . This convert 4K videos to 1080p. Is there a way to upload video recordings to Google photos automatically?

ccrisan, 2018-06-08:
There are no plans to add Google Photos uploads feature in the near future. Pull requests are welcome, though.

darkplatinum, 2018-06-08:
What do you mean by pull requests? I see the pull requests bar on GitHub. Is this used to submit requests?

Sorry I'm fairly new to GitHub so bear with me.

jasaw, 2018-06-08:
@darkplatinum Pull Request is for submitting patches back to motionEyeOS.
You'll need to fork motionEyeOS project on Github, create a feature branch, commit your changes to this feature branch, then you'll be able to create a pull-request to submit your feature branch changes.

darkplatinum, 2018-06-09:
@jasaw thanks for the info.

closed 2018-06-09 by darkplatinum