motioneye-project/motioneyeos #1538: Detect on dark field

alexei-developer, 2018-06-08:
Hello, why detect angle on dark field?

RPi3
meye-7ffdc66b
motionEye Version | 0.39
Motion Version | 4.1.1
OS Version | motionEyeOS 20180401
[image]

jasaw, 2018-06-08:
Do you mean to ask why the red motion detection box goes into the blacked out area?
That's because the motion box is the smallest bounding rectangle that covers all the areas that have "motion" and it can only draw one rectangle. "motion" is just pixel intensity change (after despeckle filter if enable).
In your case, you have a T shaped detection zone, so if there's any "motion" at two of the edges of the T zone, you're going to get a red rectangle going into the blacked out area.

Hope that answers your question.

alexei-developer, 2018-06-10:
Hello, top angle cannot be on dark field =) After experiments this sitatution exist only first 10 seconds after click button "Apply". Continue work without this problem =)

closed 2018-06-10 by alexei-developer