motioneye-project/motioneyeos #1541: Motioneye ssh stuck after password

RickvanHek, 2018-06-09:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: Can't see it (because i can't access the webpage) but it's the latest.

### Board Model

I am using the following board/model: Raspberry Pi 3 Model B+

### Camera

I am using the following type of camera: MJPEG Camera

My camera model is: Logitech Webcam C200

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet through a Powerline Adapter

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

None

### Log Files

I consider the following log files relevant to this issue:

Can't access logs.

I can't access my motioneye webpage through wifi, it just keeps loading. When I try to ssh to it and input as username: admin, i get the message "Welcome to ..." and then i have to enter my password. I do so and it just doesn't do anything. After a while i get "Server unexpectedly closed network connection".

It used to work, but now I can't do anything. Can't load the webpage, can't load with ssh. Any fixes?

jasaw, 2018-06-12:
@RickvanHek 
Questions:
- Are you able to try without the powerline adapter? Connect the Pi straight to your switch via ethernet.
- Can you keep pinging the Pi and see how many percent of packets are lost?

RickvanHek, 2018-06-23:
@jasaw 
Yes directly it has the same problem.
I can't ping to my pi, it saids:
"Ping request could not find host xx.xxx.xx.xx:8080. Please check the name and try again." 
i use 8080 as port.