motioneye-project/motioneyeos #1543: pizero w problem with more camera and "eme" 20180602

tigernero79, 2018-06-10:
I'm testing last nigtly of June 2 and I'm noticing the problem of the round circle icon with camera as attached that nn allows you to see the original picamera on pi zero w. I'm practically forced to remove power and reconnect it to get an image from the camera and camera but after a while the problem is repeated. I'm forced to go back to the stable version, it's a shame because I used to see the videos from the interface was comfortable
[image]

ccrisan, 2018-06-20:
@tigernero79 we have an issue template. Please use it and provide all the required information so that we can figure out what's going on.

tigernero79, 2018-06-21:
unfortunately I returned to the stable release where everything works, obviously without the ability to see the videos directly from the web. the problem is the non visualization of the image after some time of the pi plus camera zero w. I do not know if the same happens on pi3. forced to switch off and on again. the bug release is 20180602 the April one was fine but had problems with video viewing via the web. I came back to the building to have no problems.

ccrisan, 2018-06-21:
@tigernero79 I would really like to help you get this solved but I need some log files at least. If you have the time to test the new version again (maybe on a spare Pi Zero board), just let me know.

roelfsre, 2018-09-03:
Dear CCrisan,

I have the same issue with my Raspberry Pi zero W. 
After a while the connection to the camera is lost. I can still go to the website, but it will show the message: connection to the camera is lost.
I use putty to SSH into the Pi and reboot, than it is OK again for a while.
I also have a Raspberry pi 2 (regular one) and on this Pi I do not have the issue.
Both Pi's are on MotionEye 20180627
Attached the log-files from the Pi Zero.
If you need more information, please let me know.

Regards,

René

[motioneye.log](https://github.com/ccrisan/motioneyeos/files/2346383/motioneye.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/2346384/motion.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/2346385/messages.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/2346386/dmesg.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/2346387/boot.log)

closed 2019-08-29 by tigernero79