motioneye-project/motioneyeos #1545: Update from 20180101 to 20180602 not working

surak, 2018-06-11:
MotionEyeOS detects an update, applies it, says the update was successful, but the version remains the same. Rebooting doesn't help. 

from the log:

2018-06-11 14:47:14: [motioneye]     INFO: updating to version 20180602...
2018-06-11 14:47:16: [motioneye]     INFO: stopping motioneye watch script
2018-06-11 14:47:16: [motioneye]     INFO: stopping netwatch script
2018-06-11 14:47:16: [motioneye]     INFO: downloading firmware version 20180602
2018-06-11 14:47:38: [motioneye]     INFO: extracting firmware
2018-06-11 14:47:38: [motioneye]     INFO: flashing boot partition
2018-06-11 14:47:39: [motioneye]    ERROR: Exception in callback <functools.partial object at 0xb4b6e4b0>
Traceback (most recent call last):
  File "usr/lib/python2.7/site-packages/tornado/ioloop.py", line 605, in _run_callback
  File "usr/lib/python2.7/site-packages/tornado/stack_context.py", line 277, in null_wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/platformupdate.py", line 58, in perform_update
Exception: firmware flash boot failed

### motionEyeOS Version

I am running motionEyeOS version: 20180101

### Board Model

I am using the following board/model: Raspberry Pi Zero

### Camera

I am using the following type of camera: Pi NoIR

### Network Connection

My motionEyeOS unit is connected to the network via: Wifi

### Peripherals

USB Wifi Adapter EDIMAX

ccrisan, 2018-06-11:
Try updating to 20180224 first, using `fwupdate upgrade 20180224`.

surak, 2018-06-11:
Works, thanks.

closed 2018-06-11 by surak