motioneye-project/motioneyeos #1548: Low framerate on Raspberry Pi W

boraygowda, 2018-06-13:
Hi All,
I've seen a couple of threads that talk about low framerate, but when asked for more debug information no further information was supplied. Well, I have the same issue and I have compiled the details that were requested earlier.

My setup is a Raspberry Pi W with the Pi camera V1.3.
* Build info:
_motionEye Version | 0.39.1_
_Motion Version | 4.1.1_
_motionEyeOS | 20180602_

* Recording is H.264/OMX, recording only when motion is detected.
* Quality is set to 1200x800 at 25 fps.
* Fast Network streaming is Off.

I've noticed 100%  CPU usage on the system (screenshot attached).
The Pi-W has heatsinks and is well ventilated.  _vcgencmd measure_temp_ reports temp between 60 and 70 degrees C.

Attached screenshots:
FFProbe output
top output

config and log files attached:
thread-1.conf:    https://pastebin.com/K4RCF4Cw
motioneye.log:   https://pastebin.com/9ux4wRK8
motion.log:   https://pastebin.com/XmLxQm0K

Hope this information helps identify my low framerate issue.
Thanks for the awesome work on motionEyeOS and thank you for your time with this help.

Thanks.
BG.

jasaw, 2018-06-13:
Try this Pi Zero W configuration:
* Resolution: 1280 x 720
* Frame rate: 20 fps
* No live-stream and close your web browser connection as it loads the CPU.

boraygowda, 2018-06-13:
Hi Jasaw,
Thanks so much for responding and for your recommendations.
I've set the parameters exactly like you've mentioned.  Closed the browser and disabled streaming.
The CPU usage is a constant 70% now, with the process using most CPU being motion executable.
I've also blocked the camera to simulate no motion, but despite seeing a black/blank image the CPU usage is still 70%.
Ran a quick test of motion detection and ran ffprobe on the resultant mp4.  Results are around the same or worse (frame rate is 1.54fps and bit rate is about 180kbps):
https://ibb.co/gv3PLJ

What is the recommended settings for motion detection? Somehow I can bring the CPU usage a little more and increase the video quality?

Also, with H.264/OMX the GPU is being used for sure, right?

Thank you!
-BG.

boraygowda, 2018-06-13:
I turned off still images and got a significant boost in video quality.
Frame rate is up to 12.7fps and bit rate is 2000 kbps.

closed 2018-06-13 by boraygowda

jasaw, 2018-06-14:
Yes, H.264/OMX uses the GPU to encode video, but also relies on the CPU to feed it with images. If GPU is staved because of a choked CPU or limited memory bandwidth, you're going to end up with low frame rate. I recommend watching the CPU usage **when** motion is triggered because it's the most CPU intensive at this time. Make sure you stay below 80% CPU usage at all times. Enabling features like light-switch detection, despeckle filter, drawing motion debug box, etc, also put more load on the CPU. You'll have to tune the config to get the right balance.

boraygowda, 2018-06-14:
Thank you. I'm playing around with the parameters and will pick the optimal one.

Thanks so much for your helpful suggestions and time.
-BG.