motioneye-project/motioneyeos #1550: Connection Refused - front-end not loading on Raspberry Pi Zero W

billysyrett, 2018-06-15:
### Preliminary Docs

- [x] I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
- [x] I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20180224 (but have also tried 20180401, nightly-dev, 20180602)

### Board Model

I am using the following board/model: Raspberry Pi Zero W

### Camera

I am trying to set up a fast network camera.

My camera model is: [Standard Raspberry Pi Camera v2.1](https://shop.pimoroni.com/products/raspberry-pi-camera-module-v2-1-with-mount?variant=19833929735)

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Log Files

Cannot get log files due to connection issue
 
### Issue
I currently have 1 Raspberry Pi 3B+ running as a hub. I am trying to connect a Raspberry Pi Zero W as a camera.  motionEyeOS seems to install fine on the Zero without any error messages but refuses connections to port 80.  This means I cannot configure the camera or any other settings (such as setting the device up as a fast network camera).  The device _does_ allow connections to port 8081 to view the camera stream.

The hub device works absolutely fine and both devices have official power supplies so don't believe power is an issue.

[Screenshot of hub front-end refusing to connect](https://i.imgur.com/m7PUsFq.png)

[Screenshot from Google Chome refusing to connect](https://i.imgur.com/5lMQ4pu.png)

[Screenshot of working :8081 connection](https://i.imgur.com/1tn2iu0.png)

ccrisan, 2018-06-18:
Connection refused on port 80 could be caused by one of the following (as far as I can tell):

 * you're not using motionEyeOS but rather motionEye for your actual camera system(s) and that's port 8765 by default
 * the motionEyeOS setup has somehow got corrupted and it simply won't start correctly anymore (in which case you'll need to reflash everything)
 * you have manually changed port 80 to something else in Expert Settings
 * you've got the wrong IP address or you have some other network connectivity issues

mike-o-tronic, 2019-04-30:
motionEyeOS Version
20190413

Board Model
Raspberry Pi Zero W

Log Files
Had weird character lines with /00/00/00... in between reboots. Don't know if that's supposed to be there, but the editor definitely complained about this.

Issue
Web interface was at port 8765 instead of 80 and the system kept rebooting every 5 minutes or so.

And in case someone does stupid like me the solution is:
Reflash the sdcard, save the wpa_supplicant.conf on the boot **partition**  with the other files (bootcode.bin etc) not in the empty boot folder and voilà!