motioneye-project/motioneyeos #1551: Not detecting external USB Hardrive

Gawron10001, 2018-06-15:
Hub: RPI3 B+
Cameras: 3x rpi0 W + rpi camera v2

My hub device isn't detecting external hdd in enclosure, i took hdd and put it in the enclosure, it is beeing detected by my pc and laptop and the hub is detecting usb stick for storage. The hdd is like 300gigs. 

Soft
rpi3 b+ :20180401
rpi0: 20180401

alpa2009, 2018-06-20:
I seem to have the same problem...    I can see an external 750GB USB disk with ldisk -l as
/dev/sda1        2048 1465092095 1465090048 698.6G  7 HPFS/NTFS/exFAT
 , but it is not visible in the File Storage - Storage Device pull down list in the GUI interface...

ccrisan, 2018-06-20:
@alpa2009 what do the commands `df` and `mount` say? Can you attach your dmesg.log file here?

alpa2009, 2018-06-21:
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/2122457/dmesg.log)

attached...

df:

Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root               178482    170967         0 100% /
devtmpfs                378780         0    378780   0% /dev
tmpfs                   383388        16    383372   0% /dev/shm
tmpfs                   383388       740    382648   0% /tmp
/dev/mmcblk0p1           20422      8962     11460  44% /boot
/dev/mmcblk0p3        30343536    249900  28529196   1% /data
/dev/mmcblk0p3        30343536    249900  28529196   1% /home/ftp/sdcard
/dev/mmcblk0p3        30343536    249900  28529196   1% /home/ftp/storage

mount:
/dev/root on / type ext4 (ro,relatime,block_validity,delalloc,nojournal_checksum,norecovery,barrier,user_xattr,acl)
devtmpfs on /dev type devtmpfs (rw,relatime,size=378780k,nr_inodes=94695,mode=755)
proc on /proc type proc (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /dev/shm type tmpfs (rw,relatime,mode=777)
tmpfs on /tmp type tmpfs (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
/dev/mmcblk0p1 on /boot type vfat (ro,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p3 on /data type ext4 (rw,noatime,data=ordered)
/dev/mmcblk0p3 on /home/ftp/sdcard type ext4 (rw,noatime,data=ordered)
/dev/mmcblk0p3 on /home/ftp/storage type ext4 (rw,noatime,data=ordered)

ccrisan, 2018-06-21:
@alpa2009 indeed the partition never gets mounted. Normally the following [udev rule](https://github.com/ccrisan/motioneyeos/blob/master/board/common/overlay/etc/udev/rules.d/61-usb-storage.rules) should take care of it automatically:

    SUBSYSTEMS=="scsi", ENV{ID_FS_TYPE}=="ntfs" RUN+="/bin/mkdir -p /data/media/%k", RUN+="/usr/bin/ntfs-3g /dev/%k /data/media/%k"

Could you please run these commands manually and see if they actually work?

    /bin/mkdir -p /data/media/sda1
    /usr/bin/ntfs-3g /dev/sda1 /data/media/sda1

alpa2009, 2018-06-21:
2nd command fails:
/usr/bin/ntfs-3g /dev/sda1 /data/media/sda1
NTFS signature is missing.
Failed to mount '/dev/sda1': Invalid argument
The device '/dev/sda1' doesn't seem to have a valid NTFS.
Maybe the wrong device is used? Or the whole disk instead of a
partition (e.g. /dev/sda, not /dev/sda1)? Or the other way around?

should I format in NTFS?  I think it's exFAT now...

ccrisan, 2018-06-21:
@alpa2009 I have no experience with exFAT but from what I can see, it's not supported by the ntfs-3g mount utility so you'll probably have to use one of the supported filesystems (fat32, ntfs or ext4).

alpa2009, 2018-06-21:
problem solved...   disk should be formatted in supported filesystem.   Thanks Calin!

closed 2021-04-16 by Gawron10001