motioneye-project/motioneyeos #1554: Request timed out

Jakethetechy, 2018-06-18:
I am attempting to set up a camera system with Raspberry Pi Zero W devices as cameras, connected via a wireless network to my Debian machine which will act as a hub. I can remotely access all cameras via their IP addresses, but when I try to add them to the server i recieve the "Request timed out" error.

I have tried changing the port number and have tried using another Pi Zero as the server with the same result. Help would be much appreciated.

I am running motionEyeOS version: 20180602
I am running MotionEye version 0.39.1 on the Debian machine

I am using the following board/model: Raspberry Pi Zero W and an iMac late 2011 running Debian 9

I am using the following type of camera: Raspberry Pi camera

My motionEyeOS unit is connected to the network via: Wep/Wpa2 WiFi

I am not using any peripherals other than a DELL monitor via HDMI

I consider the following log files relevant to this issue:

[dmesg.log](https://github.com/ccrisan/motioneyeos/files/2113034/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/2113035/messages.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/2113036/motioneye.log)
[streameye.log](https://github.com/ccrisan/motioneyeos/files/2113042/streameye.log)

ccrisan, 2018-06-18:
Please attach a screenshot of the timeout message that you get when you try to add your cameras to the hub.

Jakethetechy, 2018-06-18:
Attached below is the screenshot when I try to add a remote motioneye camera. I have found that adding it as a network camera does work and that is how I will use it for now, connecting to the individual cameras if I need to.

[image]

ccrisan, 2018-06-19:
Remote motionEye cameras are usually on port 80 (motionEyeOS)  or 8765 (motionEye).

raspberrycoulis, 2018-07-24:
@ccrisan I'm now getting this after updating to `20180627`. Already had remote motioneyeOS cameras connected, but now getting request timed out.

I changed the default port from 80 to 8080 and 8086 on the cameras I've been using (without issue) until recently, so doubt the port number has anything to do with is as I've matched the same port in the remote camera - i.e. `http://192.168.0.1:8086`.

Any ideas what is happening?

raspberrycoulis, 2018-07-24:
@ccrisan I've just upgraded to `dev20180712` and it now works again. Hope this helps.