motioneye-project/motioneyeos #1557: Call discord webhook?

Bluscream, 2018-06-19:
I wonder how one would call a Discord webhook with motionEyeOS. I have set up the webhook as
[image]
and took https://discordapp.com/developers/docs/resources/webhook#execute-webhook as query parameters as explained in
[image]
so in the end i have
[image]
```
https://discordapp.com/api/webhooks/458659855943598110/qQaJ54FfeZ77cKfdxbmm8cmUec_cXfmHZkKyCBb9hrNiqUIKyXiUSmxo1FnNFRptPaXO?content=Motion%20detected!%20Frame:%20%q%20Event:%20%v
```

IAmOrion, 2018-06-22:
I'm confused... are you stating what you've done, or are you asking a question?
Are you saying you tried this webhook but it didn't work?

Bluscream, 2018-06-22:
> Are you saying you tried this webhook but it didn't work?

Exactly and i have no idea why, except that i might have not understood how this setting works in meOS

IAmOrion, 2018-06-22:
Gotchya, ok, tell me the expected results, and what you setup in Discord and I'll try and replicate it and see if I can make it work

Bluscream, 2018-06-22:
> and what you setup in Discord

That's visible on the screenshots in post 1, it's not more than that.

> tell me the expected results

[image]

IAmOrion, 2018-06-22:
Have you checked this: https://support.discordapp.com/hc/en-us/articles/228383668

IAmOrion, 2018-06-22:
If it helps, you could use the "Run An End Command" to run a python script that posts to discord using your webhook.  

I tested that already and it worked :) 

You could include that date and time of sending in the message, and potentially (depending on your setup) include the last captured still image.

Would you be interested in that?

Bluscream, 2018-06-22:
Of course but im not sure if motioneyeos supports python

Bluscream, 2018-06-22:
Also i hopy python is not too resource intensive for the pie

IAmOrion, 2018-06-22:
MotionEye is pretty much BUILT on python lol.  Almost everything it does is based on a python script!  MotionEyeOS just encapsulates MotionEye really.

Look here: https://github.com/ccrisan/motioneye/tree/master/motioneye

Everything is a python script - albeit in the compiled release, the python scripts are compiled (eg .pyc instead of .py)

It's not intensive at all - I currently use a python script I wrote, that takes the still image, submits it to openalpr to see if there's a car in the photo, if there is, it gets the plate via OCR, returns the vehicle details, then sends all the data, -- eg, time, date, photo, message via pushover so I get a push notification on my iPhone :)

Bluscream, 2018-06-22:
How do i get that onto the pi? Just use WinSCP?

IAmOrion, 2018-06-22:
@Bluscream yes, WinSCP should do it.  I personally have a mac so I used Transmit but any SFTP client will do it.... in your case, yes WinSCP should be fine :)

Bluscream, 2018-06-22:
Do i only need the discordpy file and whats the ssh password ^=^

IAmOrion, 2018-06-22:
You only need the discord.py file yes.

The password is whatever you set as your username and password for admin of motioneyeos

Bluscream, 2018-06-23:
THere is no /data sadly

```[root@CCTV data]# ls
2018-06-21/   lastsnap.jpg@ lost+found/   output/
etc/          log/          media/        varlib/
[root@CCTV data]#
```

IAmOrion, 2018-06-23:
Hi @Bluscream 

1)  Firstly, you were already in your /data folder

[image]

You only had to type "cd /data/" and it would also have taken you into your data folder.

2) No - this has nothing to do with startup commands!? I don't mean to sound disrespectful now, BUT, **I already told you with a screenshot** where you need to add the command to run this script.  If you run it directly from terminal it WILL only run once.  You need to add the script to either "Run A Command"  OR "Run An End Command".  Then, whenever your camera detects motion, it will run that script.

Here is the screenshot again...

[image]

With your opening question being about webhooks, I presumed you had at least some mediocre knowledge in stuff like this... But then you didn't even realise you were already in your /data folder so I'm guessing you've NEVER really used terminal, or done anything like this?

3) Times are a little odd in Discord, if you do a google search you will see hundreds of posts of people complaining times are off...

That being said, did you check the script and set your timezone? By default it's "Europe/London" because I'm in the UK.  You may need to change this.  

Also, this script doesn't make motioneye take a photo, it simply uses the last photo taken.  So if you ran this from terminal as a test, it will just use the last taken photo - so if your last motion event was 3 hours ago, then the last picture is 3 hours.  So if you run this script from terminal, the picture it sends will be 3 hours old hence the reason your picture timestamp is different to the time the notification was sent.

It's worth noting too, that the script is actioned would be triggered when a motion event happens... So it's normal to see anywhere from a few seconds difference to a minute maybe, depending on your config

4) That means there's no picture to send on the second camera.  Make sure it has taken at least one picture - you can manually take a picture within motion eye if needed.  Then test again.

**5) I'm going to repeat this -- You MUST add this script to the "Run A Command" OR "Run An End Command" (NOT BOTH!!!) found under "Motion Notifications".  That way, it will push the message to Discord everytime there is a motion event.**

Bluscream, 2018-06-23:
4)

[image]

IAmOrion, 2018-06-23:
Interesting... 

Can you use WinSCP and screenshot the Camera1 folder.  I wonder if the "lastsnap.jpg" alias is missing

Bluscream, 2018-06-23:
Also can't we use a argumentparser together with
[image]
to make sure no image is sent twice?

Bluscream, 2018-06-23:
Main ENtrance

IAmOrion, 2018-06-23:
Also, you don't need to send arguments?? Time and date is sent from the script, and because you're using On End, it will only trigger once the detected motion ENDs and therefore *should* only send 1 alert.

Also, the WinSCP screenshot you sent me says "/data/output/Camera1/" but in your code you changed it to "/data/output/Camera2/" ....

Bluscream, 2018-06-23:
Ohh lol, lemme check the other folder too

IAmOrion, 2018-06-23:
I have a feeling, that if using local storage like you are... Unless you change the storage path via the settings, I believe it always uses "/data/output/Camera1/"  regardless of what you actually name the Camera

Bluscream, 2018-06-23:
```
[root@CCTV data]# find . -print | sort | sed 's;[^/]*/;|---;g;s;---|; |;g'
.
|---.keep
|---2018-06-21
| |---22-10-42.mp4
| |---22-10-42.mp4.thumb
| |---22-12-54.jpg
| |---22-27-26.mp4
| |---22-27-26.mp4.thumb
| |---22-32-01.mp4
| |---22-32-01.mp4.thumb
|---cheat.py
|---discord_main.py
|---discord_net.py
|---etc
| |---2.zip
| |---adjtime
| |---crontabs
| |---date.conf
| |---hostname
| |---localtime
| |---monitor_1
| |---motion.conf
| |---motioneye.conf
| |---ntp.conf
| |---os.conf
| |---prefs.json
| |---proftpd.conf
| |---raspimjpeg.conf
| |---shadow
| |---shadow-
| |---smb.conf
| |---ssh_host_dsa_key
| |---ssh_host_dsa_key.pub
| |---ssh_host_ecdsa_key
| |---ssh_host_ecdsa_key.pub
| |---ssh_host_ed25519_key
| |---ssh_host_ed25519_key.pub
| |---ssh_host_rsa_key
| |---ssh_host_rsa_key.pub
| |---static_ip.conf
| |---streameye.conf
| |---tasks.pickle
| |---thread-1.conf
| |---thread-1.conf.bak
| |---thread-2.conf
| |---thread-2.conf.bak
| |---version
| |---watch.conf
| |---wpa_supplicant.conf
|---lastsnap.jpg
|---log
| |---boot.log
| |---dmesg.log
| |---messages
| |---motion.log
| |---motioneye.log
| |---samba
| | |---cores
| | | |---nmbd
| | | |---smbd
| | |---log.nmbd
| | |---log.smbd
| |---wpa_supplicant.log
| |---wtmp
|---lost+found
|---media
| |---motioneye_192_168_2_1___sysadmin
| |---motioneye_192_168_2_1_jmicron_generic_sysadmin
| |---motioneye_fritz_nas_jmicron_generic
| |---motioneye_fritz_nas_jmicron_generic_sysadmin
|---output
| |---.keep
| |---Camera1
| | |---.keep
| | |---2018-06-21
| | | |---22-16-18.jpg
| | |---lastsnap.jpg
| |---Camera2
|---paths.py
|---varlib
| |---dhcp
| | |---dhclient.leases
| |---logrotate.status
| |---samba
| | |---account_policy.tdb
| | |---group_mapping.tdb
| | |---private
| | | |---msg.sock
| | | | |---1003
| | | | |---1006
| | | | |---1007
| | | | |---1010
| | | | |---1011
| | | | |---1013
| | | | |---1014
| | | | |---1015
| | | | |---1016
| | | | |---1036
| | | | |---1037
| | | | |---1040
| | | |---passdb.tdb
| | | |---secrets.tdb
| | |---registry.tdb
| | |---share_info.tdb

```

Bluscream, 2018-06-23:
> I have a feeling, that if using local storage like you are... Unless you change the storage path via the settings, I believe it always uses "/data/output/Camera1/" regardless of what you actually name the Camera

Isn't that a bug then? :O

IAmOrion, 2018-06-23:
See.... /data/output/Camera2 is an empty folder.  /data/output/Camera1 has the lastsnap.jpg file... so your script needs to be `FILENAME = '/data/output/Camera1/lastsnap.jpg'` NOT ../Camera2/..

Bluscream, 2018-06-23:
Can you edit the script slightly so it reads the timezone from the meOS config file and says "no image" if the found image is older than 1 minute?

IAmOrion, 2018-06-23:
> Isn't that a bug then? :O

Not necessarily... since by default, when you ADD a camera, it IS "Camera 1".  You can then change the name of the camera.  I imagine it's not worth the coding to update the storage path along with the camera name, because someone might name their camera "This is my front garden! Using IP Address 192.168.0.1" or something stupid.  Hence why you have Storage Options

IAmOrion, 2018-06-23:
> Can you edit the script slightly so it reads the timezone from the meOS config file and says "no image" if the found image is older than 1 minute?

Possibly - I tried to keep the scripts as easy as possible but also "stand-alone" to a certain extent.  Purely because for most it would be easier to alter settings all in one place instead of some settings being in the script and others coming from elsewhere, or motioneye where they may have set incorrectly, or simply not bothered to set but then don't know where to look.

I'll look into updating it though.

IAmOrion, 2018-06-23:
> What do these mean and what should i choose to make sure nothing is mixed?

Are you using 2 cameras connected to the same Pi ?

Bluscream, 2018-06-23:
Of course. Anything wrong with that? :confused:

IAmOrion, 2018-06-23:
> Of course. Anything wrong with that? 😕

Absolutely not, no problem at all.... I was just asking to confirm the setup.  I have my setup as 4 Pi Zero W's, all using storage on a Pi3 hub.  So I can view all 4 camera from a central point.  Each Pi has it's own camera... 

If you were using 1 camera per Pi, and had 2 Pi's the output folder wouldn't mix anyways as it's local storage...

HOWEVER, since you ARE using 2 cameras on 1 pi, set Storage Device to Custom Path exactly like your second screenshot.  But, set 1 to use `/data/output/Camera1` and the other to use `/data/output/Camera2`

Bluscream, 2018-06-23:
Ahh okay, thanks will try! what do these other mean though? The part1 and part3 ?

IAmOrion, 2018-06-23:
> Ahh okay, thanks will try! what do these other mean though? The part1 and part3 ?

Make sure to update the script to reflect the folder paths accordingly.  Eg, the one I just told you to set to camera1 you will want to change back to camera2 in the pathname lol.  - it should work once the storage device path has been updated.

Part 1 & Part 3 are to do with your SDCard and/or an attached USB device perhaps

Bluscream, 2018-06-23:
> Make sure to update the script to reflect the folder paths accordingly. Eg, the one I just told you to set to camera1 you will want to change back to camera2 in the pathname lol. - it should work once the storage device path has been updated.

I changed everything thanks. What will happen if still images are set to manual and there's no image available, can it send the webhook with the "no image" image or without any embeds?

> Part 1 & Part 3 are to do with your SDCard and/or an attached USB device perhaps

Ahh thanks for the heads-up!

IAmOrion, 2018-06-23:
> Isn't that a bit of a overkill? with 4 pi's you could already do mining

Because I am covering 4 different locations, eg Front Drive way, Rear Drive way, Garden etc.  It's far easier to use a single Pi Zero W with a Pi Camera than it would be trying to run 4 100+meter usb cables each place I want a camera lol . WiFi cameras for the win!!  

Besides, in order for any real results from Pi mining i'd have to own like 20,000 of them to produce any real earnings :|

> I changed everything thanks. What will happen if still images are set to manual and there's no image available, can it send the webhook with the "no image" image or without any embeds?

Well, because the script uses the Alias that motioneye itself created (lastsnap.jpg) it will always send the last photo taken.  Of course, there has to be at least 1 photo for the lastsnap.jpg alias to work.

There's no checking to see if a file exists, I made this as SIMPLE script.  In order to cover all grounds / all bases as they say it's a fair amount of work.... and since I was only doing this to help and not for the masses (in other words, I'm doing it for free) I didn't bother with additional checks like that.  It's assumed that you're using motion events with still images on... otherwise, what's the point of using this to send a picture on motion events, if you're not taking any pictures? lol

Bluscream, 2018-06-23:
Seems like the timezone is in a directory

[image]

Bluscream, 2018-06-23:
> what's the point of using this to send a picture on motion events, if you're not taking any pictures? lol

To inform me that there was motion and maybe check. I just don't have a lot of space as you saw in the screenshots

IAmOrion, 2018-06-23:
> To inform me that there was motion and maybe check. I just don't have a lot of space as you saw in the screenshots

Well, ok, I mean I could updated the script with an option I guess.  eg SEND_PIC = 1 or 0.  so if set to 0 it just send a message and not a pic.  Is your setup available externally via a web address?  You could include a URL in the message instead of a pic perhaps.

Bluscream, 2018-06-23:
> I could updated the script with an option I guess

Thanks, i'm already working on the timezone parsing and argument parser

> Is your setup available externally via a web address? 

Half-half i'm using a VPN to access LAN from WAN

IAmOrion, 2018-06-23:
> Thanks, i'm already working on the timezone parsing and argument parser

Please make sure you share any alteration / updates here for me to see so at a later date I can perhaps incorporate them into a more robust complete script.

Not sure if you looked at my github repo where you downloaded this script, but I've been working on a few add-on scripts for MotionEyeOS.

Bluscream, 2018-06-23:
Can you send me you noimage.jpg ?

Bluscream, 2018-06-23:
> Please make sure you share any alteration 

The timezone part is done although very crappy but i'm sure that stuff won't change anytime soon in meOS

```py
TIMEZONE = os.path.realpath('/data/etc/localtime').replace('/usr/share/zoneinfo/posix/','')
```

Bluscream, 2018-06-23:
> Interesting! I'm looking at this myself, but I was going to simply tap into motioneyeos's settings and pull it from there lol

That was my way too, or do you mean the settings on the webinterface? Look at the `motioneye.conf` file
```
local_time_file /data/etc/localtime
```
[image]

IAmOrion, 2018-06-23:
> That was my way too, or do you mean the settings on the webinterface? Look at the motioneye.conf file

I was going to look into grabbing the same way the sendmail script does or something similar (from the motioneye code) however, that's not actually going to work, so I think the way you're doing it is actually going to be the way to go

Bluscream, 2018-06-23:
i could have done some path.join but as the OS is unix only it won't matter

IAmOrion, 2018-06-23:
Nah, the way you've done it is the same thing I was going to do.  I wanted to use the same method as the motioneye scripts do, but it's far easier to do what you've done

IAmOrion, 2018-06-23:
`	if os.path.isfile(FILENAME):
		c.setopt(c.HTTPPOST, [("payload_json", json.dumps(WEBHOOK_CONTENT)),("file", (c.FORM_FILE, FILENAME, )),])
	else:
		c.setopt(c.HTTPPOST, [("payload_json", json.dumps(WEBHOOK_CONTENT)),])`

That will check if a file exists and if not, won't try and send

Bluscream, 2018-06-23:
Sorry for removing all your comments but it was a bit bulky to work with (just for me others will probably appreciate it)

```py
import pycurl, cStringIO, glob, os, json, pytz, datetime, time, argparse

class motionEyeDiscordWebHook(object):
	WEBHOOK_URL="https://canary.discordapp.com/api/webhooks/458658245289050112/j-gFfa8x592zzaR4xz0aLl9Ao0y5bIpyosBqMPtf9uIio89xxkpsC5eFoXPvwnN-zzBw"
	WEBHOOK_TITLE = "__**Motion Detected!**__\n\n"
	WEBHOOK_MESSAGE = " - Motion was detected on main01"
	TIMEZONE = os.path.realpath('/data/etc/localtime').replace('/usr/share/zoneinfo/posix/','')
	MESSAGE_DATEFORMAT = "%d/%m/%Y - %H:%M:%S"
	SHARE_FOLDER = '/data/output/Camera2/'
	DEBUG = True
	utc = pytz.timezone('UTC')
	now = utc.localize(datetime.datetime.utcnow())
	la = pytz.timezone(TIMEZONE)
	local_time = now.astimezone(la)
	CURRENT_TIME = local_time.strftime('%Y-%m-%dT%H:%M:%S.000Z')
	MESSAGE_TIME = local_time.strftime(MESSAGE_DATEFORMAT)
	WEBHOOK_CONTENT = { "content": WEBHOOK_TITLE + MESSAGE_TIME + WEBHOOK_MESSAGE,"embeds": [{"timestamp": CURRENT_TIME }] }

	def get_latest_file(self):
		TODAYS_DATE = datetime.datetime.today().strftime('%d-%m-%Y')
		TODAYS_DATE_FOLDER = self.SHARE_FOLDER + TODAYS_DATE + '/*.jpg'
		LIST_OF_FILES = glob.glob(TODAYS_DATE_FOLDER)
		try:
			return max(LIST_OF_FILES, key=os.path.getctime)
		except ValueError:
			return '/data/output/noimage.jpg'

	def send_to_discord(self, FILENAME=None):
		buf = cStringIO.StringIO()
		c = pycurl.Curl()
		c.setopt(c.URL, self.WEBHOOK_URL)
		c.setopt(c.WRITEFUNCTION, buf.write)
		c.setopt(c.HTTPHEADER, ["Content-Type: multipart/form-data"])
		c.setopt(c.USERAGENT, "MotionEyeOS")
		if os.path.isfile(FILENAME):
			c.setopt(c.HTTPPOST, [("payload_json", json.dumps(self.WEBHOOK_CONTENT)), ("file", (c.FORM_FILE, FILENAME,))])
		else:
			c.setopt(c.HTTPPOST, [("payload_json", json.dumps(self.WEBHOOK_CONTENT))])
		c.setopt(c.VERBOSE, self.DEBUG)
		c.perform()
		c.close()
		f = buf.getvalue()
		buf.close()
		if self.DEBUG: print(f)

if __name__ == '__main__':
	wh = motionEyeDiscordWebHook()
	wh.send_to_discord(wh.get_latest_file())
```

IAmOrion, 2018-06-23:
> Sorry for removing all your comments but it was a bit bulky to work with (just for me others will probably appreciate it)

No problem - I designed my script to be as user friendly as possible for newbies too - you've "complicated" things there making this less user friendly so to speak from a "newbie needs to edit" point of view.  But is perfect for your own usage of course

> now let's visit the configparser

What do you want to parse so to speak? The only thing thing motion eye gives that you can't already do really is frame number.... and that's kind of pointless and not needed really. (IMO)

Bluscream, 2018-06-23:
```
class motionEyeDiscordWebHook(object):
	parser = argparse.ArgumentParser()
	parser.add_argument("url", help="Discord Webhook URL", type=str)
	parser.add_argument("--name", help="Name of the camera", type=str, default="Camera")
	parser.add_argument("--path", help="Path of the camera output", type=str, default="/data/output/Camera1/")
	args = parser.parse_args()
```

Bluscream, 2018-06-23:
> What do you want to parse so to speak? The only thing thing motion eye gives that you can't already do really is frame number.... and that's kind of pointless and not needed really. (IMO)

but you can set some stuff in the commandline that's not a variable

IAmOrion, 2018-06-23:
I see! I avoided for that for one reason.... the text box where you enter the chosen command to run is tiny, so some people may struggle so to speak figuring out all the correct command line arguments to parse, hence why I built this as a self contained script.  However, doing what you're doing is obviously far better from a "more professional" and singular script perspective as you obv wouldn't need multiple copies of the same script, you just use different arguments :)

IAmOrion, 2018-06-23:
You're making me want to alter my github repo, so that I have "basic" scripts and "advanced" scripts ha.  Basic would be all self contained, advanced would use arguments.

It's just about trying to make something that everyone can use without too much of a headache, and I know from experience scripts that require arguments are often harder for novices to get to grips with lol

Bluscream, 2018-06-23:
```
import pycurl, cStringIO, glob, os, json, pytz, datetime, time, argparse

class motionEyeDiscordWebHook(object):
	parser = argparse.ArgumentParser()
	parser.add_argument("url", help="Discord Webhook URL", type=str)
	parser.add_argument("-n", "--name", help="Name of the camera", type=str, default="Camera")
	parser.add_argument("-p", "--path", help="Path of the camera output", type=str, default="/data/output/Camera1/")
	parser.add_argument("--debug", help="Prints debug infos to console", type=bool, default=False)
	parser.add_argument("-t", "--time", help="%d/%m/%Y - %H:%M:%S from motionEyeOS".replace("%","percent_"), type=str)
	parser.add_argument("-q", "--frame", help="%q from motionEyeOS".replace("%","percent_"), type=int)
	args = parser.parse_args()
	MESSAGE = " - Motion was detected on %s" % args.name

	def get_latest_file(self):
		TODAYS_DATE = datetime.datetime.today().strftime('%d-%m-%Y')
		TODAYS_DATE_FOLDER = self.args.path + TODAYS_DATE + '/*.jpg'
		LIST_OF_FILES = glob.glob(TODAYS_DATE_FOLDER)
		try:
			return max(LIST_OF_FILES, key=os.path.getctime)
		except ValueError:
			return ""# '/data/output/noimage.jpg'

	def send_to_discord(self, FILENAME=""):
		buf = cStringIO.StringIO()
		c = pycurl.Curl()
		c.setopt(c.URL, self.args.url)
		c.setopt(c.WRITEFUNCTION, buf.write)
		c.setopt(c.HTTPHEADER, ["Content-Type: multipart/form-data"])
		c.setopt(c.USERAGENT, "MotionEyeOS")
		utc = pytz.timezone('UTC')
		now = utc.localize(datetime.datetime.utcnow())
		la = pytz.timezone(os.path.realpath('/data/etc/localtime').replace('/usr/share/zoneinfo/posix/',''))
		local_time = now.astimezone(la)
		CURRENT_TIME = local_time.strftime('%Y-%m-%dT%H:%M:%S.000Z')
		MESSAGE_TIME = local_time.strftime("%d/%m/%Y - %H:%M:%S")
		if self.args.time: MESSAGE_TIME += " (meOS time: %s)"%self.args.time
		if self.args.frame: MESSAGE_TIME += " [Frame: %s]"%self.args.frame
		if os.path.isfile(FILENAME):
			c.setopt(c.HTTPPOST, [("payload_json", json.dumps({ "content": MESSAGE_TIME + self.MESSAGE,"embeds": [{"timestamp": CURRENT_TIME }] })), ("file", (c.FORM_FILE, FILENAME,))])
		else:
			c.setopt(c.HTTPPOST, [("payload_json", json.dumps({ "content": MESSAGE_TIME + self.MESSAGE }))])
		c.setopt(c.VERBOSE, self.args.debug)
		c.perform()
		c.close()
		f = buf.getvalue()
		buf.close()
		if self.args.debug: print(f)

if __name__ == '__main__':
	wh = motionEyeDiscordWebHook()
	wh.send_to_discord(wh.get_latest_file())
```

Bluscream, 2018-06-23:
> It's just about trying to make something that everyone can use without too much of a headache, and I know from experience scripts that require arguments are often harder for novices to get to grips with lol

You are true about that hehe, but i find it more convinient to have one script for both cams instead having one for each

Bluscream, 2018-06-23:
This will be the finished command

```
python /data/discord_net.py https://canary.discordapp.com/api/webhooks/458658245289050112/j-gFfa8x592zzaR4xz0aLl9Ao0y5bIpyosBqMPtf9uIio89xxkpsC5eFoXPvwnN-zzBw -n "Main" -p /data/output/Camera1/ -q %q -t "%d/%m/%Y - %H:%M:%S" &
```

IAmOrion, 2018-06-23:
> You are true about that hehe, but i find it more convinient to have one script for both cams instead having one for each

Absolutely, you're 100% right, and an all encompassing singular script is far better, but as mentioned novices (from years of experience) really struggle with having to parse arguments, and prefer to easily edit in script.

With that said, I'm going to make "advanced" versions of my scripts on my repo to support args.  Not because I need to, but because really I shouldn't be so lazy haha

Bluscream, 2018-06-23:
Of course everything but the url is optional

IAmOrion, 2018-06-23:
> Of course everything but the url is optional

Obviously lol

Bluscream, 2018-06-23:
Thanks for your help with this :)

[31 more replies]

closed 2018-06-23 by Bluscream