motioneye-project/motioneyeos #1558: Pi Zero unstable, frequently can't view image

llamafilm, 2018-06-20:
motionEyeOS version 20180602
Pi Zero W
Pi Cam NoIR v2.1 setup as MMAL

I'm setting this up for the first time, and the feature set looks awesome, but it's very unstable so far.  Sometimes the web UI works fine, but sometimes it shows the gray camera symbol with a line through it.  When this problem happens, I get dozens of these lines repeated in the log:
`[motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED`
What does this mean?

I have disabled SMB, FTP, and motion detection for now, until I can get it running well.  Running `free -m` shows that I have around 200MB free memory.  I also get randomly disconnected from ssh sometimes with no log messages.  Do you think the CPU simply can't handle this high resolution?  I started out streaming 720p30, then I lowered it to 12fps, but I'm not sure if that helped or not.  I have not done any overclocking.

Another strange thing is that I always see this line at boot, even though SMB is disabled:
`[motioneye]     INFO: smb mounts started`

### Log Files
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/2117883/motioneye.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/2117884/motion.log)

tigernero79, 2018-06-20:
same problem I reported here:

https://github.com/ccrisan/motioneyeos/issues/1543

I returned to the stable release of the 20180224
and the problem does not happen anymore, only that we can not exploit the video playback directly from the web as it is a function added later

ccrisan, 2018-06-20:
@llamafilm:

1. Do you use SMB (a network share on your network) to store your videos?
2. When you say "I have disabled SMB", do you mean the SMB server on your motionEyeOS unit? If yes, that doesn't have anything to do with the SMB *client* that is used by motionEye to store your media on some NAS in your network.

In any case, we have an issue template that you seem to have ignored. Please use it or at least provide all the required information.

llamafilm, 2018-06-20:
I am not using any network share.  I am not even recording video, only streaming for now.

I actually followed your template exactly, I just left out the the headings.  I am not using any peripherals, and the network connection is obviously Wifi.  What other information should I provide?

ccrisan, 2018-06-20:
@llamafilm ok, fair enough. Have you ever got it stable, given your workarounds? If not, could you try reducing the framerate and/or resolution and see if the problem persists?

Another idea is to enable debugging (in Expert Settings) and try to render it unstable again. Then attach here the logs (motion.log, dmegs.log, messages.log and boot.log are of particular interest).

llamafilm, 2018-06-21:
Sorry, I don't think I'm going to have time to follow up on this issue.  I realized that what I really need is an h.264 streaming box, so I'll be looking at other software options.  Thanks for the reply.

closed 2018-06-21 by llamafilm