motioneye-project/motioneyeos #1560: Continuous Recording - Grey for first few seconds up to a minute.

IAmOrion, 2018-06-22:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

YES

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

YES

### motionEyeOS Version

motionEye Version | 0.39.1
Motion Version | 4.1.1
OS Version | motionEyeOS 20180602

### Board Model

RPi Zero W

### Camera

MMAL

My camera model is: Official Raspberry Camera v2.1, also tested with waveshare camera, Zimix IR Cut camera and usb logitech c310

### Network Connection

WiFi

### Peripherals

N/A

### Log Files

N/A

I am using "Continuous Recording", all other services are OFF.  I am using the H264/OMX codec, set to 900 seconds (15 mins) max movie length.  I have so tested with numerous values up to 3600 (1 hour)

[image]

The videos are being created as expected, and works really well with one exception...

Each video that get created, there's a grey "screen" - eg, blank recoding, for anywhere from 5 seconds up to 1 full minute before I see the actual camera footage.  It correctly records 15mins per video, but it of course means I am in effect losing up to 1 min in footage each time it makes a new file.

ccrisan, 2018-06-22:
@jasaw any ideas why this could happen? Could it be because of the limited power of a Zero W?
@IAmOrion can you try reducing the resolution and/or framerate and see if the problem persists?

IAmOrion, 2018-06-22:
Update: 

So I lowered my settings completely to 2 frames per second and 0% movie quality and 320x200 resolution.  

Exactly the same problem.  a 15min is only like 450kb haha - but it still has grey/blank recording for the first 

I tried to upload the video to YouTube to show you BUT youtube is refusing to process it because it's just that low quality!

ccrisan, 2018-06-23:
You might want to isolate the problem by disabling any other functionality you may be using. Then, if you enable debugging in expert settings, motion.log should contain some useful pieces of information regarding your problem. @jasaw may identify a problem by looking through that log. If he can't figure out what's going on, you should probably [report the issue to motion](https://github.com/ccrisan/motioneyeos/wiki/Report-Motion-Issue).

IAmOrion, 2018-06-23:
@ccrisan There is no other functionality going on... all services are off, no motion detection, nothing.  It's only task is to record 24/7 in 15 min intervals.

I will enable debugging and update tomorrow with the motion.log file :)

IAmOrion, 2018-06-23:
@ccrisan , @jasaw  -  By "accident" I've resolved my problem - but it's still a bug that someone should look into as I was able to replicate it on my other setups.

Long story short, before I used this one as a continuous recoding setup, I had been testing Motion Detection. I had set my Captured Before / After figures as per this screenshot

[image]

I had a very random thought, and I opened those settings  (which automatically turns ON the feature), I then set the 2 values (Captured Before / After) to 0, then set the feature OFF and Applied the settings.

The last 4 videos have been perfect... So it would seem as if somehow, those settings are still being taken into account and having an effect on the processing

ccrisan, 2018-06-24:
Please [open a motion issue](https://github.com/ccrisan/motioneyeos/wiki/Report-Motion-Issue) by provoding the `motion.conf` and the `thread-1.conf` files that cause this issue, together with the `motion.log` file generated with debugging info.

closed 2018-06-24 by ccrisan

IAmOrion, 2018-06-24:
Will do ... I *though* I'd fixed it, but no, it's still a problem for the first 5 seconds or so.  Will raise the issue on motion thanks

IAmOrion, 2018-06-24:
@ccrisan - sorry to "@" you, but I did as instructed here (as you directed):

https://github.com/ccrisan/motioneyeos/wiki/Report-Motion-Issue

However, everything works exactly as expected... there's NO blank recording for the first x number of seconds... the videos are perfect... maybe the issues isn't with Motion after all, after I rebooted and restarted motion eye, the videos have the blank frames at the beginning again... So wouldn't that imply it's potentially a motioneye thing?

jasaw, 2018-06-25:
@IAmOrion Sorry about my late response. The grey screen you're getting is because of motion.

You mentioned
>So I lowered my settings completely to 2 frames per second...
... but it still has grey/blank recording for the first 21 seconds.

And your captured before was set to 40 frames, and minimum motion frames set to 2. Max movie length of 900 seconds.

I think this is what happened in motion:
1. Configured for continuous recording, so flush all `precap` buffer (40 + 2 frames), and start recording. The `precap` buffer is empty because it is always recording, so flush empty frames which are grey images. In effect, you get 42 grey frames, at 2 fps, that's 21 seconds.
2. Recording reaches 900 seconds, so close off the video file and start a new recording by repeating step 1.

I would consider this as a motion bug.

Can you please attach your `thread-1.conf` here?

As a test, can you do a test run with a reduced max movie length? Try 60 seconds max movie length, run for exactly 120 seconds. You should get exactly 2 files, exactly 60 seconds each file. If you get (60+21) seconds each file, then it's definitely a motion bug.

IAmOrion, 2018-06-25:
Hi @jasaw , no need to apologise.

I agree with wha you said, I also think now, that perhaps it's also simply down the processing power - or rather the limitations of it, of the RPi Zero W.

The reason I say this, is that when I followed the instructions from here: https://github.com/ccrisan/motioneyeos/wiki/Report-Motion-Issue and I stopped motioneye, and then ran motion independently - it recorded flawlessly.

[image]
 
No blank starts, all perfectly 15mins, NO lost time or frames whatsoever.

However, upon rebooting, despite playing with the recording length (I did as you suggested and set to 60 seconds to test) it seems to 'struggle' depending on the file size.  If you look at this screenshot, you can see as it becomes night time - the video file size is smaller (less detail etc) and thus, the times cut perfectly in 10min slots as programmed.  But earlier in the day, larger files size (since there's more "content" to record) it struggles.  This makes me believe it's about the RPi Zero's capabilities rather than an issue.  

[image]

As you can see by the file names (the time) there are times where there's some seconds missing, and others (when smaller file size) where there's no issue

So I think my bug was 2 fold.  As you mentioned above, I agree and believe the 40 frames precap caused an issue but I resolved that by settings the values to 0.  My second problem however, I think is purely down to the spec of the Pi Zero.  I'm not sure which Codec would be best to use in terms of using less resources (storage space is 4TB so file size doesn't matter)

jasaw, 2018-06-26:
@IAmOrion The thread-1.conf file looks normal to me, I would expect Pi Zero W to handle this without any problem. A few questions:
- Where are you writing your video files to? `/data/media/motioneye_192_168_0_200_storage_sda1_front_cctv_rpizerow_cam1__admin` is your file server over the wifi network?
- Did you live video stream enabled or have a web browser connected to the Pi Zero W? Live video stream loads the CPU and high quality video congests the wifi network. The wifi radio on the Pi Zero W is pretty lousy, so you'll have to watch how much data you push through wifi too.

IAmOrion, 2018-06-26:
@jasaw 

> Where are you writing your video files to?  

Yes I am using network storage.  My setup is 4 x RPi Zero W connected to a central RPi 3 hub.  Although currently there are only 2 RPi Zeros active.  They use the network storage option to save files to an external hard drive connected to the RPi 3 Hub

[image]

> Did you live video stream enabled or have a web browser connected to the Pi Zero W?

No, live stream is disabled.  I do now and then login to the web ui as I've been playing with the setup but not often.

jasaw, 2018-06-26:
@IAmOrion 
> Smaller files, the WiFi is fine, but larger files it struggles with perhaps?

That is one hypothesis. You could try 25% video quality, with live stream disabled.