motioneye-project/motioneyeos #1563: "Accessing network share failed: cannot mount network share!" but settings are correct

daheefman, 2018-06-25:
## SETUP:
I confirm that I have read the contributing guide, the FAQ and many relevant issues.
I am running motionEyeOS version: 20180602 on a Raspberry Pi Zero W v1.1
I am using a this camera: https://www.aliexpress.com/store/product/Raspberry-Pi-Zero-Camera-Focal-Adjustable-Night-Vision-Camera-Module-2-IR-Sensor-LED-Light-for/1381141_32800015642.html
configured as a V4L2 Camera

## ISSUE:
I have been trying for days to mount my Synology DS1618+ network share for file storage.
Here is my config:
[image]

I can access and write to the share from my Windows PC by going to:
_\\192.168.1.5\SecurityFootage\deck_
using the 'admin' login with the correct password (contains letters, numbers and an !)
MotionEye web interface reports:
> "Accessing network share failed: cannot mount network share!"

I have read many posts including the following:
https://github.com/ccrisan/motioneyeos/issues/231
I following ccrisan's advice to run these SSH commands:
_mkdir -p /data/media/motioneye_192_168_1_5_securityfootage_admin_
_mount.cifs //192.168.1.5/securityfootage/deck /data/media/motioneye_192_168_1_5_securityfootage_admin -o username="admin",password="<mypassword>"
but I get the following SSH error:
_mount error(95): Operation not supported_
_Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)_
I tried adding all the possible 'sec=' parameters with no luck.
I get the same issue when I enabled the guest account on the NAS (without a password) and tried accessing as guest.

Any advice on how to get this working would be greatly appreciated. :)

### LOGS:
The motioneye.log file (attached) contains:
> 2018-06-25 13:48:13: [motioneye]    DEBUG: making sure mount point "/data/media/motioneye_192_168_1_5_securityfootage" exists
> 2018-06-25 13:48:13: [motioneye]    DEBUG: mounting "//192.168.1.5/SecurityFootage" at "/data/media/motioneye_192_168_1_5_securityfootage" (sec=None)
> 2018-06-25 13:48:13: [motioneye]    DEBUG: mounting "//192.168.1.5/SecurityFootage" at "/data/media/motioneye_192_168_1_5_securityfootage" (sec=none)
> 2018-06-25 13:48:13: [motioneye]    DEBUG: mounting "//192.168.1.5/SecurityFootage" at "/data/media/motioneye_192_168_1_5_securityfootage" (sec=ntlm)
> 2018-06-25 13:48:13: [motioneye]    DEBUG: mounting "//192.168.1.5/SecurityFootage" at "/data/media/motioneye_192_168_1_5_securityfootage" (sec=ntlmv2)
> 2018-06-25 13:48:13: [motioneye]    DEBUG: mounting "//192.168.1.5/SecurityFootage" at "/data/media/motioneye_192_168_1_5_securityfootage" (sec=ntlmv2i)
> 2018-06-25 13:48:13: [motioneye]    DEBUG: mounting "//192.168.1.5/SecurityFootage" at "/data/media/motioneye_192_168_1_5_securityfootage" (sec=ntlmsspi)
> 2018-06-25 13:48:13: [motioneye]    ERROR: failed to mount smb share "//192.168.1.5/SecurityFootage" at "/data/media/motioneye_192_168_1_5_securityfootage"
 
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/2131457/motioneye.log)

ccrisan, 2018-06-25:
This sounds like a duplicate of https://github.com/ccrisan/motioneyeos/issues/1501 and https://github.com/ccrisan/motioneyeos/issues/1493.

cedrik9, 2018-06-25:
If it the the same, you can test it by adding 
,vers=1.0
direcly after your password in motioneeye network share.
i test it without patch from ccrisan with 39.1 and it runs to me

daheefman, 2018-06-25:
I've tried mounting manually using the command in my original post with the vers=1.0 parameter and it still fails.
I also tried, as cedrik said, to put ,vers=1.0 in the password field after my password but this doesn't work either.

ccrisan, 2018-06-25:
@daheefman what does your command look like?

cedrik9, 2018-06-26:
Hello @daheefman,
try for share name: SecurityFootage/deck and for root direktory only /

if i connect to my fritzbox i must do it on that way

ccrisan, 2018-06-26:
@daheefman can you please try the latest [nightly dev build](https://github.com/ccrisan/motioneyeos/releases/tag/nightly-dev) and see if the problem has been fixed?

daheefman, 2018-07-03:
Hi @ccrisan, this fixed my issues, thank you!

closed 2018-07-03 by daheefman

TyLewis-ZA, 2018-07-16:
Hi @ccrisan , Firstly thanks for the amazing work on this software! I seem to be having the same issue as above that I cant change my storage device to Network Share. I am running on a Raspberry Pi Zero W with OS version motionEyeOS dev20180712. I have just updated to the latest nightly build with the same error, however with the same details I can access the share from other devices on the network.

2018-07-16 11:43:56: [motioneye]    ERROR: failed to mount smb share "//192.168.1.102/recordings" at "/data/media/motioneye_192_168_1_102_recordings_user"
2018-07-16 11:43:56: [motioneye]    ERROR: access to network share //192.168.1.102/recordings failed: cannot mount network share
2018-07-16 11:44:08: [motioneye]    ERROR: failed to mount smb share "//192.168.1.102/recordings" at "/data/media/motioneye_192_168_1_102_recordings_user"
2018-07-16 11:44:08: [motioneye]    ERROR: access to network share //192.168.1.102/recordings failed: cannot mount network share

daheefman, 2018-07-18:
Hey @TyLewis-ZA, check that you have SMB1 enabled for you share, that was part of my issue in the end.

wendy2702, 2018-07-19:
Hi guys,

i run motioneye on a old IBM Laptop but like to store my Videos/Pictures on my NAS but for some reason i didn't see the menu like you.

Used Versions:

[image]

This is how it looks for me:

[image]

Can somebody please tell me what i need to do to get the "Network Share" option displayed?

Thanks for help

ccrisan, 2018-07-19:
@wendy2702 see https://github.com/ccrisan/motioneye/wiki/Configuration-File#smb_shares. Also, make sure you run as root.

TyLewis-ZA, 2018-07-20:
Hi @daheefman, SMB1 did solve this issue, unfortunately, now I have the issue where my camera stops working from https://github.com/ccrisan/motioneyeos/issues/1582 . This might be a noob question but I am fairly new to Linux/ motioneyeos, Is there a build that uses a higher SMB version that does not cause the cameras to stop or is there a way I could manually mount the share via cli changing the SMB version?

keyanmca, 2019-05-02:
Iam facing the same issue. 

Motion Eye Log:

1970-01-01 00:03:21: [motioneye]    ERROR: failed to mount smb share "//10.10.3.26/media/Camera1" at "/tmp/motioneye_10_10_3_26_media_camera1_pi"
1970-01-01 00:03:21: [motioneye]    ERROR: access to network share //10.10.3.26/media/Camera1 failed: cannot mount network share
1970-01-01 00:03:24: [motioneye]    ERROR: failed to mount smb share "//10.10.3.26/media/Camera1" at "/tmp/motioneye_10_10_3_26_media_camera1_pi"
1970-01-01 00:03:24: [motioneye]    ERROR: access to network share //10.10.3.26/media/Camera1 failed: cannot mount network share
1970-01-01 00:03:52: [motioneye]    ERROR: failed to mount smb share "//10.10.3.26/media/Camera1" at "/tmp/motioneye_10_10_3_26_media_camera1_pi"
1970-01-01 00:03:52: [motioneye]    ERROR: access to network share //10.10.3.26/media/Camera1 failed: cannot mount network share

Kindly let me know if any thing iam missing.

Thanks

AngelofromTurin, 2020-05-26:
> Hello @daheefman,
> try for share name: SecurityFootage/deck and for root direktory only /
> 
> if i connect to my fritzbox i must do it on that way
> 
> if i read for mount error , i found that it is possible that your nas use an other smb vers.
> which one is supportet ? vers 1 or 2 or 2.1 or 3 ???
> so you have to use the right one.
> maybe it is a setting in your nas

I need to mount motion eye Storage on my FritzBox nas ... but in nubby sorry can help me ? Thank you :)

starbasessd, 2020-05-26:
There are several different versions of SMB, with different standards.
Most currently use 3.0, but some older NAS only support 2.x or 1.0
Try
Storage Device: Network Share
Network Server: fritz.nas (or the IP Address of the fritz box)
SMB Protocol Version: 3.0
Share Name:    If you have a windows pc open File Explorer, in the address
bar (says > Network > and type \\fritz.nas <enter>
     look for the fritz.nas machine., shares on the right
     use one of the share names shown (possibly: /SecurityFootage/
Share Username: whateve user you created
Share password: whatever password you created
Root Directory: whatever path you need that exists (in your case /deck)
Do a ;Apply: before doing a 'Test".
If it doesn't work, what error do you get?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AngelofromTurin, 2020-05-27:
> There are several different versions of SMB, with different standards. Most currently use 3.0, but some older NAS only support 2.x or 1.0 Try Storage Device: Network Share Network Server: fritz.nas [...]

Thank you for youre help!
Thise the error message:
Accessing network share failed: Cannot mount network share!

Put the pic of my configuration:
[image]
[image]

starbasessd, 2020-05-27:
> type \fritz.nas look for the fritz.nas machine., shares on the right use

For Server:  fritz.nas
for root path try: /FRITZ.NAS/Toshiba_Ext/homeassistant

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AngelofromTurin, 2020-05-27:
@starbasessd 
you are the only man on many pages, groups and chats who helped me on the right path !!! I'm really grateful to you!!!
now works with this configuration:
network server: fritz.nas
smb protocol: 1.0
sharen name: FRITZ.NAS/Toshiba_Ext/homeassistant
                      (shareservername/usbhddname/directory)
root: /
user/password:

I don't know why but the user was also a problem, I used a user activated specifically for other addons on homeassistant without vpn and it didn't work; I tried with another user that I use generically to do everything with fritzbox with active vpn and this user works!

I hope it is useful to everyone :)
thank you very much again!

starbasessd, 2020-05-27:
You're welcome.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

suvopam, 2024-07-19:
> Hello @daheefman, try for share name: SecurityFootage/deck and for root direktory only /
>
> if i connect to my fritzbox i must do it on that way
>
> if i read for mount error , i found that it is [...]

spot on !