motioneye-project/motioneyeos #1564: Connect an external network camera.

friulancik, 2018-06-25:
Hi, my configuration is a raspberry pi3 with the Camera Module V2 - 8 Megapixel, 1080p that works. I would like to add an external network camera connected in wifi.
The external camera is already connected in wifi with the router but I do not understand which URL I have to insert in the "Add Camera ..." window
I tried to enter the local IP address of the camera for example 192.168.1.3 or http://192.168.1.3 but it does not work.

motionEye Version | 0.39.1
Motion Version | 4.1.1
OS Version | motionEyeOS 20180602

Board Model
Raspberry Pi 3 Model B

Camera
-Camera Module V2 - 8 Megapixel, 1080p
-network camera http://www.hy-happyegg.com/works/1080p/outdoor/74.html

My motionEyeOS unit is connected to the network via WiFi.

I consider the following log files relevant to this issue:
 
[messages.log](https://github.com/ccrisan/motioneyeos/files/2134710/messages.log)
[dmesg (1).log](https://github.com/ccrisan/motioneyeos/files/2134711/dmesg.1.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/2134712/boot.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/2134713/motioneye.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/2134714/motion.log)

friulancik, 2018-06-28:
I used video streaming software (VLC) and it works correctly using this URL:
rtsp://user:password@IP:rtsp port/live/ch0

However in Motioneyeos this URL does not work.

ccrisan, 2018-06-28:
@friulancik motionEye expects the username and password in separate fields. Please remove them from the URL.

friulancik, 2018-06-28:
[image]
Thanks for your interest. Unfortunately it does not work anyway. Where am I wrong?

closed 2018-06-28 by ccrisan

friulancik, 2018-06-29:
Thanks Calin, I found the solution with the following URL
rtsp: //192.168.1.4: 554 / live / ch0
To thank you for the support I would like to make a small paypal donation, do you have a special button?

ccrisan, 2018-06-29:
@friulancik you can find a donation button at the bottom of https://github.com/ccrisan/motioneyeos. Thanks!