motioneye-project/motioneyeos #1568: run end command - how can I access to filename ?

mrRobot62, 2018-06-29:
Hi,
I'm using latest version of motioneyeos - first congratulations - cool software.
This software will be used in my house automation system to watch our private parket place. if a car arrived the cam should make a snapshot (motion detected). 

Via openALPR this image will be scanned and license plate is checked against a database. If unknown car a message is send to our mobiles.

openALPR works, ioBroker as house automation works.

Now my question: if image captures and img saved on my NAS, I have to send a web request to my iobroker instance. I want to store latest filename. This filename is a value of an state inside iobroker

How can I access onto this latest filename? For me it's important, because inside iobroker a script will scan a above state and use the filename for openALPR.

Hope it's ok to describe the behavior .

Cheers
LunaX

mrRobot62, 2018-06-30:
------------ SOLVED ----------

Ah, it's easy, I misunderstood something. I called the web hook command like this
http://xxx.yyy.0.123:8087/set/javascript.0.openALPR.input.imageName?value=**%Y-%m-%d_%H-%M-%S.jpg**

It works fine - on iobroker side i got afterwards in the state this file name: 2018-06-30_07-11-44.jpg

Cheers
LunaX

ccrisan, 2018-07-04:
I'll close the issue then.

closed 2018-07-04 by ccrisan