motioneye-project/motioneyeos #1569: Fast Network Camera live feed defaults to 2 fpm when selected to be higher

kyleser, 2018-06-29:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (20180602).

### Board Model

I am using the following board/model: (Raspberry PI Zero W with 3B+ as hub).

### Camera

I am using the following type of camera: (Fast Network Camera used as Simple MJPEG Camera).

My camera model is: (MakerFocus OV5647 night vision camera).

### Network Connection

My motionEyeOS unit is connected to the network via: (Fast network cameras via WIFI, hub connected via ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue: n/a

### Log Files

I consider the following log files relevant to this issue: will provide if requested

### Issue
I have 3 cameras each connected to a Raspberry Pi Zero W.  These are all setup as Fast network cameras connected to a Raspberry Pi 3B+.  I have them all configured to 800x600 resolution and 15 fpm on both the hub and PiZW's.  The FPM on the live feed always returns back to around 2 FPM after a very short period of time.  What do I need to do to correct this?

Thanks!

ccrisan, 2018-06-30:
@kyleser I believe your setup is not ideal from a performance point of view. The 3B+ is roughly 6 times better in terms of processing power than an Zero W. You're using 3 x 3B+ units to practically collect frames (no processing whatsoever in FNC mode) and then pass the 3 streams to a Zero W that does all the heavy lifting. Do you see the problem here?

The Zero W is also limited when it comes to the network connection. I would never use a wirelessly connected unit as a hub for multiple cameras.

At this point, I would recommend using a(nother) 3B+ unit for the hub with a wired connection.

kyleser, 2018-06-30:
Thanks for the clarification and response. I’m sorry but I feel I may have left misleading or unclear info, allow me to put back in order.  What I meant to say was I have 3 Zero W’s set up as FNC all connected to a single 3B+ wirelessly. The 3B+ (As the hub) is connected via ethernet.

So I’m wondering if the 3B+ is capable to handle each stream of 800X600 at 15fps.  I’m finding, it’s not even when I increase the memory option to 512MB and over.

As a possible solution, I’be ordered an Odroid XU4 to be my hub for the 3 Zero W’s. 

Thoughts?

Thanks again for your time and expertise here!

ccrisan, 2018-07-04:
@kyleser increasing the memory should have no significant impact. However, the Pi 3B+ should be able to handle a few 800x600@15fps streams. The bottleneck could well be the network connection (bound at ~10MB/s). I would first try to reduce the quality of the FNC streams. Mine work at about 10% quality factor and I'm fine with what I get.

ccrisan, 2018-07-04:
I meant your RPi3 B+'s network connection is limited to roughly 10MB/s and your Zero Ws may be limited even more, given their wireless connection.

kyleser, 2018-07-04:
Ah ok thanks for that.  I’m testing this out currently (changing the image quality). I plan to add 3-4 more cameras, one being Ethernet and others as wireless.  For this, I assume using the Odroid XU4 would be best (or at least better), yeah?

kyleser, 2018-07-07:
Alright so I’ve noticed changing the image quality to 10% and frame rate to 10fps seems to help keep the cameras stable. I added a forth Zero W camera and it is a tad bit further away from my router and it does just OK at the same settings.  I think the distance might effect things.

I’ve also setup a XU4 with motionEye Os with all the same settings and it performs the same.  However I’m confident the XU4 would be able to handle more streams later.  

I think what I try next is replace the further Zero W with a 3B+ (setup as wireless) and see if that performs better.

kyleser, 2018-07-13:
I’ve found Ethernet is king, especially for using RPi Zeros. I currently have 5 cameras each configured as Fast network cameras, 800x600, 15fps, and 10% imagine quality (everything else is default other than storage and static ip).  I’m useing the XU4 as a hub running motionEye OS.  I have 2 3B+ in wireless mode and 3 RPi Zero W using Ethernet (with a usb to ethernet adaptor).

I’m getting stready streams around 13-15 fps. 

I found (at least in my environment) the RPi Zero W has poor wireless communication for this setup when paired multiple cameras.

closed 2018-07-13 by kyleser

ccrisan, 2018-07-13:
That's usually the case :) Stay away from wireless connections as much as possible, when it comes to video surveillance.