motioneye-project/motioneyeos #1570: Raspberry Pi B and Zero wont update even if prerelease is set to On.

Nypan-sr, 2018-06-29:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

Nypan-sr, 2018-06-29:
Tried restarting several times after chosing Apply on both my Pi's B & Zero.

ccrisan, 2018-06-29:
What version are you currently running and what version are you hoping to upgrade to?

Nypan-sr, 2018-06-30:
Pi B started to update tonight, after some 40 minutes it said Update failed.

ccrisan, 2018-07-04:
Closing since @Nypan-sr ignores my questions and the issue template.

closed 2018-07-04 by ccrisan