motioneye-project/motioneyeos #1574: GUI update failure

Chiny91, 2018-07-04:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version:  20171003 _antique and been working flawlessly_

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: Network Camera *3

My camera model is:  Hikvision various

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals

None

### Log Files

I consider the following log file relevant to this issue: a piece of motioneye.log
`2018-07-04 19:23:19: [motioneye]     INFO: updating to version 20180627...
2018-07-04 19:23:21: [motioneye]     INFO: stopping motioneye watch script
2018-07-04 19:23:21: [motioneye]     INFO: stopping netwatch script
2018-07-04 19:23:21: [motioneye]     INFO: downloading firmware version 20180627
2018-07-04 19:23:31: [motioneye]     INFO: extracting firmware
2018-07-04 19:23:31: [motioneye]     INFO: flashing boot partition
2018-07-04 19:23:32: [motioneye]    ERROR: Exception in callback <functools.partial object at 0x7428a7e0>
Traceback (most recent call last):
  File "/opt/buildroot/motioneyeos/output/raspberrypi3/target/usr/lib/python2.7/site-packages/tornado/ioloop.py", line 604, in _run_callback
  File "/opt/buildroot/motioneyeos/output/raspberrypi3/target/usr/lib/python2.7/site-packages/tornado/stack_context.py", line 275, in null_wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/platformupdate.py", line 58, in perform_update
Exception: firmware flash boot failed`

After clicking the "Check" and "Yes" (to new version), the GUI reports that "motionEye was successfully updated" but a re-check after a manual reboot, shows it was not.   **_Both_** my motionEyeOS systems (both as above) refuse to GUI update.    Drat;  unless there is a magic tweak, I'll have to reflash the SDrams and most likely have to fidget about with ssh keys again.

ccrisan, 2018-07-04:
Please update manually to 20180224 first, using `fwupdate upgrade 20180224` command. Then you'll be able to update automatically to the latest version.

Chiny91, 2018-07-05:
Yes, the `fwupdate` command did indeed allow me to update to the latest version and all is working correctly.    

The ssh keys got trashed by `fwupdate` but I guess anyone who is using them, should be capable of re-installing them 😄   At least I wrote down the commands to be taken for such a re-install, so just a short step away from an automating, Python script.

OK, I can now abandon my motionEyeOS boxen until the next "must have" fix - amazing reliability, much appreciated.

closed 2018-07-05 by Chiny91