motioneye-project/motioneyeos #1575: FW Update issue

helmar74, 2018-07-05:
Hi,

I managed to upgrade 4 of my 5 Motioneyeos Cams to Version 20180627.
The 5th cam (which is a RPI Zero W and running 20180224) failed to upgrade.
I tried manual upgrade with /sbin/fwupdate,
This failed with the following error;
./fwupdate upgrade 20180627
upgrading to 20180627
downloading...
curl: (7) Failed to connect to api.github.com port 443: Connection timed out
no such version

If I'm looking on the other cam for new versions, the same error occurs:
 ./fwupdate versions
curl: (7) Failed to connect to api.github.com port 443: Connection timed out

Thanks a lot for your great work!

Helmar

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version
20180224

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model
RPI Zero W

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

helmar74, 2018-07-05:
I think it'S a Github issue:
https://status.github.com/messages
"We're investigating reports of intermittent connection failures."

ccrisan, 2018-07-05:
Either that, or your 5th cam's network connectivity is bad.

helmar74, 2018-07-05:
No, that's not the case. After https://api.github.com is working again. The FW Upgrade went fine....
Maybe you can implement an "Exception" when the upgrade is started via WebGui and api.github.com isn't reachable.
After api.github.com issue was solved, Versions where found again:
 ./fwupdate versions
20180627
20180224
20171008
20171003
20170710
20170329
20170212
20161125
20161027
20160915
20160705
20160605
20160410

helmar74, 2018-07-05:
[image]

helmar74, 2018-07-05:
Closing issue

closed 2018-07-05 by helmar74