motioneye-project/motioneyeos #1576: network share

nataliesharpe96, 2018-07-05:
i am trying to save to a network shared folder on windows 10 home
network ip is 192.168.1.208
share name is server
login is windows 10 email
password is windows password
path is /ducks/camera1
/ducks/camera2

when i use the most recent update version released 7/1/2018 it cannot connect to share but when i use the version released about a month ago it works just fine.

so just ;letting you know that something about this most recent update doesnt like saving to a share folder on windows 10 home.

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

ccrisan, 2018-07-05:
What exact version do you use?

nataliesharpe96, 2018-07-05:
currently i am using 

OS Version | motionEyeOS 20180602

and its working just fine but version 20186027 will not let me save to a windows share folder

ccrisan, 2018-07-05:
I believe that has to do with the SMB protocol version. Could you please try and [force SMBv1](https://support.microsoft.com/en-us/help/2696547/how-to-detect-enable-and-disable-smbv1-smbv2-and-smbv3-in-windows-and) on your Windows machine?

nataliesharpe96, 2018-07-06:
unfortunately im not sure how to do that most of those settings are blocked off in windows home

marshman24, 2018-07-12:
Can you make an FTP server on your windows?

tigernero79, 2018-07-12:
quietly, I also mounted last release of July 12 and access via ftp quietly,

[image]

if you have any problems with windows 10 remember to enable unsafe guest access.

in the windows execute tab that can be called via the + R window key, type "gpedit.msc" from there to go to the left in administrative templates ---> network -> lanman workstation ---> enable unsafe guest access

[image]

check the entry enable and try again

see you

tigernero79, 2018-07-27:
the home version did not activate these options, it was enough even a professional version, then make sure that the smbv1 protocol is active, as written in another post, go to

also check that in the control panel, in programs and functionalities, in activation or deactivation of the windows functions, the entry on smbv1 is checked

Bra1nK, 2018-07-29:
I have a similar problem in that since the upgrade to the most recent version I cannot connect to my windows 2000 server share.

ccrisan, 2018-08-01:
Guys, the latest nightly-dev includes an option to manually choose SMB protocol version. Please test it and see if it fixes your problems. To upgrade to latest nightly-build, use `fwupdate upgrade nightly-dev` from command line, making sure that you have Prereleases enabled in Expert Settings.

elementarykid, 2018-08-01:
Thanks very usefyl

aptraum, 2018-08-01:
motionEyeOS dev20180730 almost resolved the issue on my end :(.  Running on a RPi3 and the non-nightly would give me a Test Share successful confirmation but as soon as it tried to actually save video the cameras would go down until I set it back to the local save.  Been running for about 10 minutes now and have multiple saves going to the correct share destination but it did eventually error out.

I'm going to go back to some of my NAS (openmediavault) forums an try a couple of things, a few users reported issues with not the SMB1 protocol itself, but the authentication method.  There's an extra options field for the share and one of the items mentioned was to try plugging   ntlm auth = yes
into that field.  The troubleshooting was for general SMB1 shares and not specific to motioneyeOS.

[motioneye (1).log](https://github.com/ccrisan/motioneyeos/files/2251238/motioneye.1.log)

aptraum, 2018-08-01:
Changing to SMB3 seems to be working on my end with RPi3 on motionEyeOS dev20180730 and a openmediavault NAS.

Thanks!!!

Bra1nK, 2018-08-02:
I'm away on holiday at the moment so will test after my return

Bra1nK, 2018-08-09:
Just upgraded to the latest nightly build, where do I find the option to manually choose the SMB protocol version ?

Bra1nK, 2018-08-09:
Managed to get my windows 2000 server share working again after rewriting the line in the /data/etc/fstab file as follows //[Windows Share] /[Mount Point] cifs vers=1.0,user=[username],pass=[password] 0 0