motioneye-project/motioneyeos #1577: Motion Detection on 2nd (Remote) Camera ?

Klemsch, 2018-07-05:
Hello together,
I already read that a RPI3 to do motion detection both on internal CSI and remote RPI
(Catalin at google groups)

I have a Remote RPI2B (option "Fast Camera") and a RPI3 (with internal CSI camera) which should do the motion detection for both cameras.
The Remote RPI is connected as a "Simple MJPEG Camera".
I see both cameras on the rpi3 browser interface, but there is only 1 motion detection menu for the internal camera of the rpi3. 

How can I activate the motion detection for the remote camera?

Regards,
Klemsch

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20180627

### Board Model

I am using the following board/model: PI 3B (and Remote PI 2B)

### Camera

I am using the following type of camera: (both internal CSI-Camera).

My camera model is: (CSI).

### Network Connection

My motionEyeOS unit is connected to the network via:  Ethernet

### Peripherals

none

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

ccrisan, 2018-07-05:
Add your RPI2 camera as a Network Camera to your RPI3, instead of as a Simple MJPEG Camera.

Klemsch, 2018-07-06:
Thanks, Calin for the hint.
I now set the url to "http://IP-Address:8081/" and Network Camera.
It is fine now.
Motioneye is a great Programm!

Regards,
Klemsch

closed 2018-07-06 by Klemsch