motioneye-project/motioneyeos #1578: Is it possible to change the ssh port from 22 to another port?

tigernero79, 2018-07-06:
As a title I wondered if it was possible to change ssh listening port of emotioneyeos from the classic 22 to a door of my choice, because if you have several ssh services accessible from outside I would like to avoid bots or for convenience I would be able to change the door.

I tried via ssh to edit the file / etc / sshd_config but says it is read only, I would not mess with the permissions is possible?

Chiny91, 2018-07-06:
You did make the partition writable first (as per [FAQ](https://github.com/ccrisan/motioneyeos/wiki/Tweaks#partition-layout)) ?

tigernero79, 2018-07-06:
I think the command is only valid for the / boot partition or is it wrong?

closed 2018-07-07 by tigernero79