motioneye-project/motioneyeos #1579: Motioneye won't start

chrisdowdell, 2018-07-06:
Hi,
I am a Pi and Linux newbie and am running Raspbian 9 (Stretch) with motionEyeOS version 0.39.2 installed on a Rapsberry Pi 3B.
Had a nightmare updating and upgrading Stretch but got there in the end!
Followed the instructions to install motioneye on top of Stretch which was equally interesting but finally got it all installed as well.

I thought it was all sorted and tried to access motioneye on port 8765 without any luck.
Having checked the status of motioneye I can see it fails to start with the following status output:

pi@raspberrypi:~ $ sudo systemctl status motioneye
motioneye.service - motionEye Server
Loaded: loaded (/etc/systemd/system/motioneye.service; enabled)
Active: failed (Result: start-limit) since Fri 2018-07-06 16:05:13 BST; 55min ago
Process: 910 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=killed, signal=SEGV)
 Main PID: 910 (code=killed, signal=SEGV)

Jul 06 16:05:13 raspberrypi systemd[1]: Unit motioneye.service entered faile....
Jul 06 16:05:13 raspberrypi systemd[1]: motioneye.service holdoff time over,....
Jul 06 16:05:13 raspberrypi systemd[1]: Stopping motionEye Server...
Jul 06 16:05:13 raspberrypi systemd[1]: Starting motionEye Server...
Jul 06 16:05:13 raspberrypi systemd[1]: motioneye.service start request repe....
Jul 06 16:05:13 raspberrypi systemd[1]: Failed to start motionEye Server.
Jul 06 16:05:13 raspberrypi systemd[1]: Unit motioneye.service entered faile....

I'm hoping that someone can help with the failed start info as having done a bit of searching can find nothing about the above issue, plus there are no log files generated either, so no joy on tha front.

Thanks

ccrisan, 2018-07-06:
What is your motion version? What does `motion -h` say?

pharqall, 2018-07-06:
Segmentation fault.
How do I resolve this and thanks for your rapid response.  :-)

pharqall, 2018-07-06:
OK I have finally resolved it myself  :-)
I just went through another round of sudo apt-get update, sudo apt-get upgrade and sudo apt-get dist-upgrade for the 5th time today and now finally all working.

closed 2018-07-06 by pharqall