motioneye-project/motioneyeos #1582: MotionEye / Motion / Cameras just stop working

IAmOrion, 2018-07-12:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue. (YES)

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue. (YES)

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

motionEye Version | 0.39.2
Motion Version | 4.1.1
OS Version | motionEyeOS 20180627

### Board Model

RPi Zero W

### Camera

MMAL 

### Network Connection

WiFi

### Peripherals

### Log Files

Very odd, but my cameras had been working fine for a while now, running and recording 24/7.

I updated the newest stable release (20180627) and now for some reason, everything "freezes".  I can access the Web UI, and via SSH, but the camera image won't show, it's no longer recording and even running reboot in ssh does nothing (well, it just hangs so to speak) meaning I have to physically unplug and re plug in the device.

I currently have 2 devices - identical RPi Zero W with the same RPi Cameras.  (As mentioned, they have been working fine for ages without trouble) ... this morning, as coincidental as it is, they BOTH stopped at 7:15am.   BOTH!! What are the chances of them both stopping within 30 seconds of each other? 

What logs would help in this instance?

xadox-1st, 2018-07-12:
Same for me.

ccrisan, 2018-07-12:
Do you save media files remotely? Please attach all of your logs, since they all might contain useful information as to why your system hangs.

xadox-1st, 2018-07-12:
Yes, I am storing the captures on SMB.
Will post logs later.

IAmOrion, 2018-07-12:
> Do you save media files remotely? Please attach all of your logs, since they all might contain useful information as to why your system hangs.

@ccrisan Yes I do, I have a 3rd Raspberry Pi (Pi 3) running as a central hub with 4TB attached hard drive.

Logs attached.  I've prepended them cam1_ and cam2_ (and appended .log to messages file)

[cam_1dmesg.log](https://github.com/ccrisan/motioneyeos/files/2188303/cam_1dmesg.log)
[cam1_boot.log](https://github.com/ccrisan/motioneyeos/files/2188304/cam1_boot.log)
[cam1_motion.log](https://github.com/ccrisan/motioneyeos/files/2188305/cam1_motion.log)
[cam1_motioneye.log](https://github.com/ccrisan/motioneyeos/files/2188306/cam1_motioneye.log)
[cam1_messages.log](https://github.com/ccrisan/motioneyeos/files/2188307/cam1_messages.log)

[cam2_boot.log](https://github.com/ccrisan/motioneyeos/files/2188309/cam2_boot.log)
[cam2_dmesg.log](https://github.com/ccrisan/motioneyeos/files/2188310/cam2_dmesg.log)
[cam2_messages.log](https://github.com/ccrisan/motioneyeos/files/2188311/cam2_messages.log)
[cam2_motion.log](https://github.com/ccrisan/motioneyeos/files/2188312/cam2_motion.log)
[cam2_motioneye.log](https://github.com/ccrisan/motioneyeos/files/2188313/cam2_motioneye.log)

xadox-1st, 2018-07-12:
Here my logs:
[boot.log](https://github.com/ccrisan/motioneyeos/files/2188903/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/2188904/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/2188905/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/2188906/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/2188907/motioneye.log)

tigernero79, 2018-07-13:
me with release 27062018 I had many problems, with release 28062018 and last pre release 12072018 everything is fine

ccrisan, 2018-07-13:
Guys, the problem here is that the SMB protocol version has recently changed. Some users are fine with the old version, while others are fine with the new one. I personally don't know what to do about it.

IAmOrion, 2018-07-14:
@ccrisan hmmm interesting.  Did the logs indicate what actually happens? Does the SMB protocol go to sleep or "crash"? My 2 cameras lasted about 55 hours this time, before they both rebooted.  At least this time they did reboot without getting stuck in limbo so that's something I guess.

Perhaps I should go back to the older version.

matucker, 2018-07-20:
I'm having the same issue since updating to motionEyeOS 20180627
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/2214414/motioneye.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/2214415/messages.log)

tigernero79, 2018-07-20:
install release 20180712

SealedJoy, 2018-07-20:
same problem here except only saving locally no smb used. Pi zero w.

ccrisan, 2018-08-01:
Guys, the latest nightly-dev includes an option to manually choose SMB protocol version. Please test it and see if it fixes your problems. To upgrade to latest nightly-build, use `fwupdate upgrade nightly-dev` from command line, making sure that you have Prereleases enabled in Expert Settings.

IAmOrion, 2018-08-07:
Thanks, will give this a try and report back

VegasIOT, 2018-08-08:
I tried the nightly build and had the same issue, I'm going to try to fall back to 0602 on my pi zero W.  I've literally replaced all the hardware. It just seems to die after a few days

matucker, 2018-08-11:
I have the Samba server turned off and I'm storing the files locally.  I did install the latest nightly and I'm  still having the same issues on my Pi Zero W's.  I didn't change the samba version because I'm storing locally and don't have the option to change samba versions unless I change it to a network share.  Looking at my logs I still see smb mounts stopping and starting, but why would they if I have the samba service turned off?

[motioneye.log](https://github.com/ccrisan/motioneyeos/files/2279750/motioneye.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/2279751/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/2279752/motion.log)

ccrisan, 2018-08-21:
@matucker your issue is then not related to SMB itself so you can safely ignore the SMB-related comments. Judging form your logs, lines like `Aug 10 18:48:44 ZeroFive user.notice wifi: disconnected` indicate that you have a connectivity issue.

IAmOrion, 2018-08-27:
I know it's been a while now, but just to report back and say no matter what SMB protocol I choose, I still have the same problem, it just seems to be random with it's timing.

Worth noting though - and this is what actually makes this a real problem for me, is that once it does freeze up so to speak - I can't reboot it remotely.  It "hangs".  Reboot and Reboot -f do absolutely nothing.  There's no way to reboot it, I've tried stopping the motioneye services etc etc before rebooting but nope - still won't work.  I have to go and pull the plug - which is a huge pita when it's supposed to be a remote cctv unit :(

tigernero79, 2018-08-27:
Idem for me, command reboot ssh not reboot pi zero also gui web is blocked

artesdev, 2018-09-17:
Same thing for me this morning. Suddenly, the GUI of motioneye show nothing, and can't login anymore. :-(

quantum8, 2018-11-01:
Not sure if I'm having a similar issue or not. I have two zero w set up as fast cameras and a 3b set as the server. Every few hours the second zero loses the camera connection. I can still log in to reboot it to get the image back. The first zero hasn't skipped a beat.

[Logs.zip](https://github.com/ccrisan/motioneyeos/files/2540478/Logs.zip)

tigernero79, 2018-11-02:
with the release 20181008 it seems that the problem has been solved at least for me on my pizero w no longer freezes.

https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20181008.img.xz

quantum8, 2018-11-02:
Just tried that @tigernero79, it still failed a few hours later.

VegasIOT, 2018-11-02:
For me I switched the wifi channel on my router, the one I was using was
very congested and all these problems  went away.

starbasessd, 2018-11-27:
I have the 'loss of camera as of' error on 20180627 on RPi3B. Happens to a USB attached camera. Doesn't matter where the files are being saved (local or network) I tried to update to dev20181126 but during install, it downloads, unpacks, then tries to apply, but runs out of space on the partition. (8GB SD Card). File system is trashed afterwards. Goes into a reboot loop. Unable to recover data. 
Re-imaged with 20180627. Booted, Went into KVM to console. Ran fwupdate upgrade dev20181126 before doing anything else. Same thing.  unable to get logs off of SD Card.

ccrisan, 2018-12-03:
@starbasessd I'm sorry for your inconvenience, but the partition layout has been recently changed. You'll need to write the new OS image from scratch. See https://github.com/ccrisan/motioneyeos/releases/tag/20181129.

jasaw, 2018-12-04:
@ccrisan I think there's a way to support fwupdate from old partition layout to the new one. The data partition is created by S00datapart if the partition does not exist.

1. We could tar up a few configuration files like `wpa_supplicant.conf` and `motioneye.conf`, put them in the `boot` partition temporarily. 
2. We then delete the data partition and reboot. On device boot up, S00datapart will create a new data partition while reserving 1GB for `boot + root`.
3. Untar the backed up config files from `boot` back into data partition and reboot.
4. Now we can upgrade to the new latest firmware using the usual process.

Re-doing the data partition could be worth it. We could use it to support doing factory reset. Factory reset + "bring up wifi AP if not configured" will greatly improve the user experience. Aim is to minimize the need for users to reflash SD card.

What do you think?

ccrisan, 2018-12-09:
@jasaw I have also tried to figure out a way to do this upgrade automatically but I believe there's no practical way to do that without losing user's data.

Backing up a few config files is indeed a good idea but I believe it's definitely not enough. Having this done automatically and effectively removing media files is going to piss off a lot of users. I think having them doing this explicitly is a better option, in general.

jasaw, 2018-12-09:
@ccrisan Yes, agree completely. We could add a checkbox next to firmware upgrade button that says "reformat data partition". We could also add a warning about losing media files. I think that should satisfy a lot of users.

jasaw, 2018-12-16:
@ccrisan If you're happy to add "reformat data partition" as a feature, I can create a new issue to keep track of future discussions.

IAmOrion, 2018-12-20:
For what it's worth - this is still plaguing me.  When it ultimately does lose the image, I can often login, change the SMB protocol version, click save... and then after about 20 seconds of it thinking, my feed appears again.

The biggest problem plaguing me atm though, is that sometimes, this doesn't fix the problem.  But somehow it prevents the Pi rebooting via terminal.

I've tried stopping the motioneye service, I've killed processes, I've stopped SMB/Samba etc etc - but no matter what I do, the "reboot" command does absolutely nothing.  I type reboot, hit enter.... and then it just sits there.... thinking I guess, because (best way I can explain) is your cursor moves to the next line... but that's it.... it does NOT have the # ready for your input.  Typing anything now has no effect.  ctrl+c has no effect (other than showing those funny characters on screen from pressing that combo).  

This is where the crux of my issues are.... I'm away from home, due to it being installed in a secure location, it has a battery backup supply - this means there's no WiFi wall socket I can kill the power with.  So ultimately I'm left without security and a non-working system until I get home in 3 days and physically unplug/plug back in the device.

Reading the conversation above, I guess at some point I will have to dismantle my entire setup and set everything up again from scratch, using the latest os releases (due to the partition change stuff).

ccrisan, 2018-12-26:
> @ccrisan If you're happy to add "reformat data partition" as a feature, I can create a new issue to keep track of future discussions.

@jasaw we could track this matter in a separate issue, why not.

jasaw, 2018-12-29:
Created "reformat data partition" issue here: https://github.com/ccrisan/motioneyeos/issues/1756

ccrisan, 2019-01-02:
Should we close this issue then? @IAmOrion has this issue been solved for you in a way or another?

IAmOrion, 2019-01-02:
@ccrisan unfortunately not - I still have the same problems.  Whilst it is "better" and sometimes when I lose the image I am able to change the SMB version and click Apply - and the image will return... but it's 50/50 so many times it makes no difference and I'm stuck with no image.  That then seems to create a secondary issue - no matter what I do I can not manually reboot the Pi.  Shutdown doesn't work.  I've tried stopping motion eye first, I've tried killing all tasks, tried killing them one at a time etc etc - so I end up in a position where my CCTV is down for a day or 2 until I can get physical access and pull the plug :( ... (The Pi is connected to a battery backup system so I can't simply use a WiFi socket to kill the power)

All that said - I haven't updated since I saw you say that in order to update you have to have access and re-write the sd card etc.

So I'm currently on this version:

motionEye Version | 0.39.3
Motion Version | 4.1.1
OS Version | motionEyeOS dev20180730

Is this, to the best of your knowledge, fixed in the newer pre-releases?  I ask because it's not a 5min job to access 1 of the cameras - so I don't want to spend 6 hours getting to it, disconnecting it, dismantling the housing, updating it, then putting it all back together to suffer the same issues.

In all fairness - whilst the loss of image due the SMB thing **IS** annoying, I wouldn't mind if I could simply reboot remotely.  But sadly, it seems impossible - for whatever reason, something prevents rebooting.  When I type reboot and press return, the cursor moves to a new line, but it's obviously waiting for something (the "command prompt" doesn't show on the new line, meaning it's not processed the command and is thinking about it / stuck)

ccrisan, 2019-01-11:
@IAmOrion I am not aware of any SMB-related fixes in newer versions, so that shouldn't be the main reason to upgrade. Nevertheless, at some point in time you'll need to upgrade, so why not now?

Also the fact that you can't reboot the system is something that needs to be investigated. That is definitely not normal and I'd need some logs to see what's going on.

tigernero79, 2019-01-11:
For my account I can tell you that the problem of not restarting from the raspberry pi terminal, in my case pi zero w has disappeared with the latest releases.

even to me it happened that the command reboot did not restart the raspberry, but simply wore the cursor and remained so for life.

in the latest releases the problem has been solved

IAmOrion, 2019-01-15:
@ccrisan ok, after being up for some time, a good week or so I think actually, it's just "died" again.  
When I viewed the Web interface, I had the "No Image" picture.  It was using SMB 2.1 at the time.  I tried my usual fix of changing the SMB to 3.0 and clicking apply.  That's been a hit and miss 'fix' as sometimes it works, sometimes it doesn't.  In this instance, it did NOT work.  Now I can't even access the web interface.

Rebooting remotely also refuses to work.

I can however access it via SSH / SFTP - so, before battle my way to yanking the plug out of the bloody thing - what log files would you like?  I'll download then via SFTP whilst it's in it's purgatory state of being useless and refusing to reboot.

Just the usual?  Eg:

boot.log
dmesg.log
messages.log
motion.log
motioneye.log

Any others?

ccrisan, 2019-01-23:
@IAmOrion I simply cannot tell what's going on in this case. Rebooting should especially never fail since the first thing it does is to kill the watchdog feeder. Is this something 100% reproducible?

ccrisan, 2019-01-24:
@IAmOrion when you find your unit in this bad state, could you please run the internal commands contained by `/sbin/reboot` one by one?

This is what `/sbin/reboot` looks like:

    #!/bin/bash

    # carry on with the script in case of error
    set +e

    # write buffers to disk
    /bin/sync

    # allow the shutdown script 20 seconds to shut down,
    # after which we stop feeding the watchdog
    (sleep 20 && /usr/bin/killall -STOP watchdog) &

    # actual reboot command
    /bin/busybox reboot

If anyone can spot a problem with this script, I'd be happy to address it.

IAmOrion, 2019-01-24:
@ccrisan Sure thing! I'll hold off updating this particular unit (I've updated a couple last night) because it will happen eventually! I'll report back when it does, thanks

ccrisan, 2019-01-27:
@ir-fuel what OS version?

ir-fuel, 2019-01-27:
20180627

ccrisan, 2019-01-27:
@ir-fuel see if 20190119 fixes this problem for you.

ir-fuel, 2019-01-30:
Installing the latest version in the prerelease list right now. It worked fine for 2 days and right now the system was blocked (no web server, no video uploads). I could SSH into the Raspberry Pi though. I'll let you know my findings.

ccrisan, 2019-01-30:
@ir-fuel please make sure you installed 20190119 and not some dev nightly build.

ir-fuel, 2019-01-30:
How do I do this? If I only allow official releases it tells me I am on the latest version, and if I toggle to prerelease it shows me 20190130 as build to install.

ir-fuel, 2019-01-30:
Ok got it:

note: do not attempt to update to this version automatically! Due to the different partition layout, you'll need to rewrite the OS image from scratch. Using the automatic update mechanism will, at best, fail with an error message.

I'll wipe my sd card and install it.

IAmOrion, 2019-01-30:
@ir-fuel https://github.com/ccrisan/motioneyeos/releases/tag/20190119

ir-fuel, 2019-01-31:
FYI It's still running since I upgraded the version. Will see if it stays like this.

ccrisan, 2019-02-02:
@IAmOrion `reboot` definitely fails due to waiting indefinitely for `sync`, in your case. Problems with syncing buffers are often linked to faulty SD cards. Can you please try a different (type of) SD card and see if the problem persists?

I can't tell if this particular problem can be caused by SMB, but you could temporarily disable SMB usage on your system and see if that's the cause. I would go with SD card change first, though.

GeoffMorg, 2019-02-03:
FWIW I am noob to the Pi and this is only my second project. I am having the same problem. I set it up earlier today and was successfully sending images to my NAS (using SMB 2.0) for a few hours but decided to rebuild and relocate it, and since then (and further rebuilds) the camera just drifts off after a while - and I've not even got to the point of using my NAS, so it's all local.

I'm using it headless, just accessing through the browser and I can report that the power off button works OK.

I have tried three different SD cards (brand and size) and it continues to be a problem, I have another new Zero here so will try that tomorrow.

I'm using motioneyeos-raspberrypi-20190119 and would be happy to assist with any logs as required.

btw having said that it looks like a perfect product  with the ability to do just what I need.

GeoffMorg, 2019-02-06:
Further to my post above, I had only successfully built a working ca,era 1 time out of maybe 15 times. So I took everything really slowly and documented each step with the idea that I might isolate which particular piece of config was breaking it,  And of course it worked. It worked for a full day then I powered off overnight, and at next power on I have no camera.

I did backup the config so I will re-create the card with the software, and if that is working just reload my config.

Joseph-Muc, 2019-04-10:
dear friends of motioneyeos,
same Problem here with exact the same configuration.
Running a view days very nice and find it a very usefull system.
a lot of /bin/sync lines after ps command (Stat D).
There is no possibility to kill them.

greetings
Joseph

ccrisan, 2019-04-12:
@Joseph-Muc `sync` commands are there to make sure your changed data is written to the SD card as soon as possible during boot, so that in the event of a power loss, you don't lose your configuration.

It is expected not to be able to kill it but it's definitely not normal that it runs for long periods of time. Please try replacing the SD card with a new/faster one and see if the problem persists.

C4PT41ND34DP00L, 2019-08-16:
I am having an issue similar to this.

MotionEye version is 20190427

It was working then had it powered down for a bit. I plugged it back in and connected to the web interface but the raspberry pi camera just spins and never loads.

[image]

[motion.log](https://github.com/ccrisan/motioneyeos/files/3509733/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3509734/motioneye.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/3509735/messages.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/3509736/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/3509737/dmesg.log)

starbasessd, 2019-08-16:
Please confirm your OS, browser, etc. If Chrome browser version ...100,
confirm it works in another browser. Also, do a change user in the Chrome
browser even if you change back to admin, and confirm you get black windows
for the camera(s)

-- 
Thanks

Kevin Shumaker

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

C4PT41ND34DP00L, 2019-08-16:
@starbasessd  Thank you for the information. It works in Firefox. Was not aware of an issue with Chrome.

starbasessd, 2019-08-16:
Yes, it seems to have been a Chrome issue, popping up in mobile, too.  on
my Android Chrome ...111 works, but Win10 & Android Chrome 100 and the
previous version are broken still. More than likely, a Chrome update broke
it.

-- 
Thanks

Kevin Shumaker

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ccrisan, 2019-08-19:
Fixed in recent nightly builds.

closed 2019-08-19 by ccrisan