motioneye-project/motioneyeos #1588: motioneye os won't update from 0.37.1

SixBytesUnder, 2018-07-17:
I logged in to my motionEyeOS today, clicked on a  "check" button to see if there is a software update. But when I try to update, it takes a minute then shows me a **success** message, but noting changes. I'm still on 0.37.1 and motionEyeOS 20171008 I have a lot of free space and my motioneye.log file shows following error:

	2018-07-17 20:44:05: [motioneye]     INFO: hello! this is motionEye server 0.37.1
	2018-07-17 20:44:08: [motioneye]     INFO: cleanup started
	2018-07-17 20:44:08: [motioneye]     INFO: wsswitch started
	2018-07-17 20:44:08: [motioneye]     INFO: tasks started
	2018-07-17 20:44:08: [motioneye]     INFO: mjpg client garbage collector started
	2018-07-17 20:44:08: [motioneye]     INFO: smb mounts started
	2018-07-17 20:44:08: [motioneye]     INFO: server started
	2018-07-17 20:44:58: [motioneye]     INFO: updating to version 20180627...
	2018-07-17 20:45:00: [motioneye]     INFO: stopping motioneye watch script
	2018-07-17 20:45:00: [motioneye]     INFO: stopping netwatch script
	2018-07-17 20:45:00: [motioneye]     INFO: downloading firmware version 20180627
	2018-07-17 20:47:22: [motioneye]     INFO: extracting firmware
	2018-07-17 20:47:22: [motioneye]     INFO: flashing boot partition
	2018-07-17 20:47:23: [motioneye]    ERROR: Exception in callback <functools.partial object at 0x755d0570>
	Traceback (most recent call last):
	  File "/opt/buildroot/motioneyeos/output/raspberrypi3/target/usr/lib/python2.7/site-packages/tornado/ioloop.py", line 604, in _run_callback
	  File "/opt/buildroot/motioneyeos/output/raspberrypi3/target/usr/lib/python2.7/site-packages/tornado/stack_context.py", line 275, in null_wrapper
	  File "/usr/lib/python2.7/site-packages/motioneye/platformupdate.py", line 58, in perform_update
	Exception: firmware flash boot failed
	2018-07-17 20:47:23: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081

The camera itself works just fine. A minute before it did detect movement, recorded a short video and uploaded to Dropbox. All seems normal, just update doesn't work.
I did disable "Video device" and "Motion Detection", but it didn't help. I did reboot for good measure.
What am I doing wrong?

### motionEyeOS Version

I am running motionEyeOS version: 20171008

### Board Model

I am using the following board/model: Raspberry PI 3 Model B

### Camera

I am using the following type of camera: Simple MJPEG Camera

My camera model is: Raspberry Pi v2.1 8 MP 1080p Camera Module

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

VelJP, 2018-07-27:
Hello SixBytesUnder,

I have the same problem, and after long search, find an old topic which resolve the problem.
I was stuck from MotionEyeOS 20180101...

Look this topic : 
https://github.com/ccrisan/motioneyeos/issues/1527

Solution is : 
- Activate SSH on your webGui of MotionEyeOS
- Access in SSH with login/password
- type "fwupdate upgrade 20180101"
- should download, extract, and reboot successfully.

Do the same with "fwupdate upgrade 20180224" and after "fwupdate upgrade 20180401".
After that, you should use the WebPage update button.

I hope this could help you.

ccrisan, 2018-07-27:
@VelJP is right. That's the way to update from older versions.

closed 2018-07-27 by ccrisan

jasaw, 2018-07-27:
FYI, this firmware upgrade issue is also addressed in our [troubleshooting wiki](https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#firmware-upgrade-failure).

SixBytesUnder, 2018-07-27:
Don't know why it didn't come up when I was googling for it.
Thank you for responding, I'm all up to date now.

Trefle10, 2020-08-31:
Trying to update from  20190911 version using the web interface and nothing happen...
so I tried to so it with the command

fwupdate upgrade 20200606

Result: 
upgrading to 20200606
downloading...
not enough disk space

What can I do next??

starbasessd, 2020-08-31:
If you have any videos still on the system, or jpegs, clear them off.
You can check the amount of free space with
df -h
You can backup the camera setups with the 'Backup' button, then re-image
with a newer version, then do the 'Restore'.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Trefle10, 2020-08-31:
Thanks but how to delete and where?  My storage is now on NAS but it was for a time local so that may be it.

Guess the backup/reinstall latest/restore config is the way to go.

starbasessd, 2020-08-31:
If you can get to the console (connect monitor and keyboard) or ssh into
it, you can then do a
find -name *.jpeg
or
find -name *.(whatever video format you are using)
to find the directories you were using before, and delete them.
The backup/restore will probably be fastest.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?