motioneye-project/motioneyeos #1591: Webserver not responding

MinePro120, 2018-07-20:
Info :
Logitech C27 USB webcam
Raspberry Pi 3
Wifi connection
v.20180712

Issue : So I logged in normally for the first time and applied the changes I made. It worked fine and I left it running for a couple of days. Note that I've installed heatsinks and a fan to my rpi3. I logged in as admin afterwards and the camera image was static. It was an image taken before or during the login. After a few seconds the camera "failed" and the "no camera" icon appeared. Though, the camera's light remained green, which means that the camera is working properly. I refreshed the page only to find that the webserver was not responding (err dns unreachable, or connection timeout). After a couple of attempts, it appeared to be loading (takin forever) without showing the login page. So can you help? The rpi was still working.

MinePro120, 2018-07-20:
Mysteriously, it happens when I login with my android phone.

DungeonKeeper1, 2018-08-25:
Hello, 

same problem here with 20180730 on Pi3 B. 
First connection with Android no problem. 
After setup (sure correctly) no connection to MotionEye. 

Android Firex says: Connection interrupted 

After Scan with Fing, the IP from the Raspi is not visible. 

But i have full access with all my Windows computers via Firefox. 

Is there a solution? 

Thank you very much! :)

Boksy, 2018-08-26:
Seen the same, should pull logs. Did you do this already?

MinePro120, 2018-08-26:
No logs have been uploaded yet Boksy.