motioneye-project/motioneyeos #1594: banana pi m2 ultra

nataliesharpe96, 2018-07-23:
hey just wondering what the exact banana pi model you support

all thats listed is banana pi but not a model like the raspberry pi 3 3+ and so on and theres like 5 models of the banana pi. i specifically have the m2 ultra since its quad core 2 ghz and 2gb of ram and im trying to run 3-4 cameras

ccrisan, 2018-07-27:
@nataliesharpe96 you're right: the Banana PI board variant must be specified - it's Banana PI M1. I've updated the Supported Devices wiki page.

Kronos9247, 2018-12-27:
Why does nobody on the internet have build MotionEyeOS for the Banana PI BPI-M2 Zero

ccrisan, 2019-01-02:
@Kronos9247 https://github.com/ccrisan/motioneyeos/wiki/FAQ#can-motioneyeos-be-ported-to-insert-your-board-here