motioneye-project/motioneyeos #1597: Whole Odroid setup information

stdin-stdout, 2018-07-26:
### motionEyeOS Version

I am running motionEyeOS version: 20180627

### Board Model

I am using the following board/model: Raspberry Pi Zero W

### Camera

I am using the following type of camera: Pi 5MP camera with built-in cable

My camera model is: Pi 5MP camera with built-in cable

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

stdin-stdout, 2018-07-27:
Thank you for the _MotionEye_ software

I have setup _MotionEye_ OS on a Raspbery Pi 3 Zero W with the current 5MP standard Pi camera with built-in cable. I monitor _MotionEye_ on a desktop running Debian  GNU/Linux 9. It works well except for the common sluggish frame rate and occasional WiFi disconnection.

Having read the post "_RPi 3 + motionEye = slow preview/recording fps #227_", I note that C. Crisan uses "_an Odroid XU4 running motionEye as a hub that performs all the motion detection and movie encoding_" with "_RPi1 devices just as fast network cameras, all of them working in FNA mode._"

I wonder if the whole Odroid setup could be detailed in a complete step by step installation help page. In particular I would like to know the following.

-1- What is meant by "_RPi1_" devices? Are these Raspberry Pi 1 devices or later version of the Raspberry connected to an early version of the RPi camera?
-2- I suppose "_FNA_" means Fast Network Adapter. Is this an adapter with a male USB plug on one side and an Ethernet female socket on the other side?
-3- Are both Odroid and RPis running MotionEye OS? Are the RPIs running MotionEye OS with the Fast Network camera option turned on? Or are the RPis running something else since they don't need motion detection software altogether in this setup?
-4- Is the Odroid XU4 better suited than the Odroid C2?
-5- Are all the cameras connected to the Odroid through an Ethernet switch?
-6- Among the many brands of SBCs and excluding x86 boards, is there an even better board than the Odroid or Raspberry for the hub?

Finally and before buying an Odroid board, I am testing my desktop running Debian as a hub for my single Raspberry Pi 3 Zero W based camera. I have followed the Debian installation instructions but when I try to open Firefox with http://[my_ip]:8765/ my browser isn't able to connect to that address. Any idea why? _MotionEye_ is running though, as shown by a $ps command line.

ccrisan, 2018-07-27:
Please use our [Google Group](http://groups.google.com/group/motioneye) for this kind of questions. This is not a valid issue. Nevertheless, [this wiki article](https://github.com/ccrisan/motioneyeos/wiki/Usage-Scenarios) may help you.

closed 2018-07-27 by ccrisan