motioneye-project/motioneyeos #1598: odroid xu4

nataliesharpe96, 2018-07-27:
so i have tried every last release for the xu4 to try and save the footage directly to my windows 10 shared drive, and on every last release for the odroid it fails to connect/ save to the server . not sure whats going on just says access denied even theough the credentials are correct. im hoping to have this fixed asap as i only have 30 days from 7/27/2018 to return the odroid if this fails to start working. if you can get this working i will gladly make a donation.

ccrisan, 2018-07-27:
@nataliesharpe96 whatever you do, don't return it - it's an amazing board :)

This sounds like a bug with the SMB version that's affecting all motionEyeOS builds. We're planning on adding an option to select it manually and we're hoping it will fix all recent network share-related issues.

tigernero79, 2018-07-27:
As I wrote in this post the solution is simple:
https://github.com/ccrisan/motioneyeos/issues/1576

just go :
in the windows execute tab that can be called via the + R window key, type "gpedit.msc" from there to go to the left in administrative templates ---> network -> lanman workstation ---> enable unsafe guest access

[image]

tigernero79, 2018-07-27:
also updates to the July 12 release, more stable release on my pi zero, via ssh
giving this command

"fwupdate upgrade https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-odroidxu4-dev20180712.img.xz"

nataliesharpe96, 2018-08-01:
hey the nightly update that allows you to select the smb version manually completely solved the issue. V 1 didn't allow me to save but then selecting V2 works perfectly. please make this a permanent addition to the user interface.

nataliesharpe96, 2018-08-01:
so it does save the files but they are all corrupted

nataliesharpe96, 2018-08-01:
new discovery the version select worked and the only file format the corrupts is h.264, the mpeg4 on version 2 to windows 10 home shared drive works perfectly. so please allow version select menue for smb amazing feature

jasaw, 2018-08-01:
How exactly is h264 file corrupted? What software did you use to playback?

nataliesharpe96, 2018-08-03:
i use the windows movie player for playback. when i go to watch it it tries to load it, but it cant and it just skips to the end. im taking it that means it shows as a playable video but the file is corrupted. when i switch it from h.264 to mpeg4 winows has no isue playing the video.

jasaw, 2018-08-03:
@nataliesharpe96 Try [Media Player Classic](https://mpc-hc.org/).