motioneye-project/motioneyeos #1600: Mass storage not recognized

elementarykid, 2018-07-28:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20180602

### Board Model

I am using the following board/model:Raspberry pi 0w connected with fast networking to Raspberry pi 3B+ the mass storage hub

### Camera

I am using the following type of camera: Fast Network Camera

My camera model is:Raspberry pi camera v1.3

### Network Connection

My motionEyeOS unit is connected to the network via:WiFi for 0w Ethernet and wifi for 3B+

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
 3Tb External Hard Disk

### Log Files

I consider no log files relevant to this issue:

ccrisan, 2018-07-28:
What filesystem?

elementarykid, 2018-07-28:
What do you mean

guthrik, 2018-07-28:
???????I've not raised a case ... even though I got an email saying that I did!I downloaded the package a couple of days ago but I've not even installed it yet and I certainly don't have any mass storage attached.

ccrisan, 2018-07-29:
@elementarykid I mean what filesystem do you have on your external drive (mass storage) that you say is not recognized. Example choices would be fat32, ntfs, ext4.

@guthrik what the hell?

elementarykid, 2018-07-29:
@guthrik BOI!!!

elementarykid, 2018-07-29:
Fat 16 a.k.a MS-DOS

ccrisan, 2018-07-29:
@elementarykid Fat16 is not supported by motionEyeOS. Please use one of the formats mentioned above.

closed 2018-07-29 by ccrisan

elementarykid, 2018-07-29:
@ccrisan Thanks

Burgy22, 2018-10-03:
Hi, i´ve tried using the following formats, without sucess: fat32, ntfs, ext4.
Using a Pi3B+ as Hub.
The HDD1Tb does not get recognised?

ccrisan, 2018-10-03:
@Burgy22 please attach your log files (notably dmesg.log) after plugging in your HDD.

Burgy22, 2018-10-04:
A new issue was openen #1667 and has now been solved, the issue closed.
@ccrisan thanks for this great piece of work, please reclose this issue, i´m regret for the mixup i caused.