motioneye-project/motioneyeos #1602: fwupdate doesn't appear to locate the nightly

aptraum, 2018-07-30:
More of an FYI than anything, figured I'd try the newest nightly to see if it fixed the SMB issue that appears to have been identified in several of my searches.  When I issue the fwupdate upgrade nightly-dev command I receive the following:
upgrading to nightly-dev
downloading...
no such version

Current version = 20180627
In the repository I do see motioneyeos-raspberrypi3-dev20180730.img.xz

No worries on my end though, I've been testing your awesome software out for the past week and will just reflash.  Thanks for all the work you have done to make this OS happen!

ccrisan, 2018-07-31:
@aptraum thanks for testing our nightly build. Do you have Prereleases enabled in your Expert Settings?

aptraum, 2018-08-01:
Wow, I can't believe I missed that setting.  Thanks and close at your convenience filing it in the "Another End User Error" bucket.

Thanks for the response and once again thanks for bringing this OS to the masses!

closed 2018-08-01 by aptraum