motioneye-project/motioneyeos #1603: Server crashing

nopuk, 2018-07-31:
I am running Motion eye OS, on remote camera and one hub/server to view and manage. The server has repeated just crashed for want of a better word, the Pi's lights are on including the network light, my router shows it as a connected device but I am unable to call up the web interface page or ssh into it. Not really sure how to investigate this further on my own.
( motion eye on the remote camera pi is perfect has given zero issues , only issue it logging onto the server webpage)

I am running motionEyeOS version: 
motionEye Version | 0.39.2
-- | --
Motion Version | 4.1.1
OS Version | motionEyeOS 20180627
### Board Model
raspberry pi 3 b+
### Network Connection
My motionEyeOS unit is connected to the network via ethernet
### Log Files
I consider the following log files relevant to this issue ( 14:27 is around the time of the crash and me plugging in/out power supply )
[issues.txt](https://github.com/ccrisan/motioneyeos/files/2245706/issues.txt)

Boksy, 2018-08-26:
I've seen this on my raspberry pi b 2 plus. I suspect web server fails. Can you try to putty into it?

nopuk, 2018-09-02:
Hey,
Thankfully I have not seen the issue in over two months of 24/7 use, next time it happens I will try to ssh into it, but the last time I was unable to.