motioneye-project/motioneyeos #1604: How I fixed the "starting wpa_supplicant: failed (brcmfmac)" problem

mooblie, 2018-07-31:
motionEye Version | 0.39.2
Motion Version | 4.1.1
OS Version | motionEyeOS 20180627

Raspberry Pi Zero W 1.1
USB Webcam Logitech C270
--------
Many have described the problem of trying to log a Pi Zero W onto a WiFi network, and getting a failure with "starting wpa_supplicant: failed (brcmfmac)" message during bootup.

I had tried all the many suggestions here and elsewhere with no luck, and finally hit on the solution (for me anyway).  When checking the "wpa_supplicant.conf" file with a text editor that shows HEX (to check there were only LF characters for line ends - not CR+LF) I noticed the double quotes surrounding the SSID and password were the complex triple-byte Unicode versions E28096 and E2809D (in HEX), not the simple single-byte ASCII version 22 (in HEX).  On changing those complex opening and closing double quotes to the simple version, all started working OK.

Where did the complex versions come from? I suspect that as some point I copied someone's "wpa_supplicant.conf" file from a forum, and their "helpful" text editor had substituted the complex opening and closing double quotes for the simple versions.  It is VERY difficult to see this on a printout, or on screen, unless you look VERY hard!  Anyway, it fixed it for me!! Hope it helps others too!

Refs:
http://www.fileformat.info/info/unicode/char/201C/index.htm
https://www.fileformat.info/info/unicode/char/201d/index.htm

N59175, 2020-01-11:
any chance I can use your file?  Or what hex editor did you use?

mooblie, 2020-01-11:
Hi. Let me see if I can find a copy for you. (Six months ago though!) Give me a day or so.....

mooblie, 2020-01-12:
HI again. Sorry, I don't have the copy of wpa_supplicant.conf that I actually used any more, but I believe I used Notepad++ on Windows for this.  See here:

https://www.raspberrypi-spy.co.uk/2017/04/manually-setting-up-pi-wifi-using-wpa_supplicant-conf/

...specifically in UNIX/OSX mode - as described in that article. Hope this helps!

mooblie, 2020-01-16:
Here it is:

[wpa_supplicant.conf.zip](https://github.com/ccrisan/motioneyeos/files/4072380/wpa_supplicant.conf.zip)

N59175, 2020-01-16:
Thanks I’ll give it a try

N59175, 2020-01-26:
That didn't work.  I finally got it to work by connecting to a RPI3 with wired network, configuring that pi to use wifi network, testing to make sure it connects, then taking the sd card from that pi and finding the wpa_supplicant file (the pi creates it when it's configured to use wifi) (it's in a misc directory I think), copy that file to the RPIzw and it works!!  must be formatting differences or something, I even tried creating the entire file in Linux which also failed....

mooblie, 2020-01-26:
Wow! Glad you got it working, but it sounds like a mystery, as to what actually was the cause of the problem.  Anyway - good job!

bartvalom, 2020-06-05:
Very observant, thanks. In my case I think it was caused by using Linux keyboard definition 'English (US, intl with dead keys).' Using 'English (US) keyboard definition inserts straight 0x22.

bartvalom, 2020-06-05:
By the way: I used Midnight Commander's F3 (view) command, followed by F4 (hex) as a hex editor.

dmadma1, 2024-04-16:
> Here it is:
> 
> [wpa_supplicant.conf.zip](https://github.com/ccrisan/motioneyeos/files/4072380/wpa_supplicant.conf.zip)

Only with that file work for me :)
Yhanks!
All that story with numerous wpa_supplicant.conf troubleshooting on web is a complete mess ...