motioneye-project/motioneyeos #1605: shutdown motionEyeos remotely via ssh (openhab)

roycie1282, 2018-08-01:
this is not so much an issue as is: is this possible with motioneyeos so im trying to shutdown my cameras from a remote computer automatically through ssh (via openhab). now I know this is not a normal linux system, but for my other RPI's i just have to edit " sudo visudo -f /etc/sudoers.d/010_pi-nopasswd " with the openhab user auths. on the system i want to access and then log in via ssh with the correct user and then from there it automatically does it. I know there is no sudo in motioneyeos so how and where would i add the user with permissions to motioneyeos. any help would be greatly appreciated. i have already logged in from the correct user via ssh to  get add the key (ECDSA) to the list of known hosts. hopefully this makes sense if not i will try to elaborate

ccrisan, 2018-08-02:
@roycie1282 you don't have users on motionEyeOS. There's only `root` and therefore you don't need sudo or anything of that sort. `admin` is an alias to `root`.

kiwi-cam, 2018-08-26:
Hi @roycie1282.  I use HomeBridge for my Home Automation and as part of that, I have it run a script when the last person leaves, or the first person arrives which disables one of my indoor cameras in MotionEyeOS.

If it's any help here's how I do this:

1. On My MotionEyeOS Raspberry Pi I have a script called /data/etc/enablesingle.sh which updates the config files with backed up versions for the required configuration.
2. The homeBridge server runs a script including this line to execute that script via SSH with public key authentication:
`ssh -i /var/homebridge/id_rsa -o StrictHostKeyChecking=no admin@192.168.21.51 '/data/etc/enablesingle.sh &> /dev/null'`

I also have the HomeBridge script create or delete a state file so I can report the status of the camera. This script runs every five minutes as a "fail safe":
```
#!/bin/sh
nc -z 192.168.21.51 8082
 
if [ $? -eq 1 ]; then
    #Port is closed
    rm -f /var/tmp/indoorcamera.state
else
    #Port is open
    touch /var/tmp/indoorcamera.state
fi
```

If you need more detail, let me know.