motioneye-project/motioneyeos #1607: some issues with new setup (lowfps - stream and record, zoom, fast network camera, movie codec)

nighter83, 2018-08-01:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 
20180627 on cams (0.39.2 MotionEye - 4.1.1 Motion)
Ubuntu 18.04 with MotionEye 0.39.2 and Motion 4.0 as central server for motion detection

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

RP 3 B+ for 2 cams
RPi Zero W for one cam
Ubuntu 18.04 on 4vCPU 4GB Memory VM on a HW Raid 5 as central Server for Motion Detection and recording (CPU in HW Host: 4core i7-4830k)

### Camera

I am using the following type of camera: V4L2 for cams

My camera model is: RP Cam 5MP and 8MP - IR and NOIR variants

### Network Connection

My motionEyeOS unit is connected to the network via: WPA2 secured 2,4Ghz WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

none

### Log Files

not sure which logfiles/configurations would make sense here
ffmpeg_output_movies off
height 960
stream_quality 75
threshold 7987
quality 85
noise_level 31
ffmpeg_output_debug_movies off
pre_capture 1
noise_tune on
smart_mask_speed 0
stream_maxrate 30
output_pictures off
stream_localhost off
ffmpeg_variable_bitrate 75
ffmpeg_video_codec mp4:h264_omx
text_changes off
movie_filename %Y-%m-%d/%H-%M-%S
auto_brightness on
stream_port 8081
rotate 180
stream_auth_method 1
lightswitch 0
framerate 25
emulate_motion off
snapshot_filename 
despeckle_filter 
snapshot_interval 0
minimum_motion_frames 20
stream_motion off
target_dir /data/output/Camera1
text_double on
post_capture 1
stream_authentication xxxxxxxxxxxxxxxxxxxxx
output_debug_pictures off
on_picture_save /usr/lib/python2.7/site-packages/motioneye/scripts/relayevent.sh "/data/etc/motioneye.conf" picture_save %t %f
on_movie_end /usr/lib/python2.7/site-packages/motioneye/scripts/relayevent.sh "/data/etc/motioneye.conf" movie_end %t %f
text_left Cam50
picture_filename 
locate_motion_style redbox
locate_motion_mode off
mmalcam_name vc.ril.camera
max_movie_time 0
on_event_end /usr/lib/python2.7/site-packages/motioneye/scripts/relayevent.sh "/data/etc/motioneye.conf" stop %t
text_right %Y-%m-%d\n%T
on_event_start /usr/lib/python2.7/site-packages/motioneye/scripts/relayevent.sh "/data/etc/motioneye.conf" start %t
event_gap 30
mask_file 
width 1280

I've setup several cams with various RP models and cams. 
At the moment one RP3B+ is running with a normal 8MP RP (V4L2)cam with the configuration above. 
I was just wondering about some topics: 

1. low streaming/recording rate on both the cam UI itself and on the central server (10-15 FPS average sometimes 22 or 26 fps)
I thought that the FPS should be higher on a P3 without any motion detection, as the FPS are similar (8-12) with my RP Zero W setup as fast network cam
2. Tried to run the RP3B+ as FNC and the FPS were even lower (4-8FPS)
 Do I need to change the cam type on the core server if I switch over to FNC?
3. Is there any way that we can zoom with a non-FNC setup?
4. I'm more or less forced to record in MPEG or MSMEPGv4 in order to have reasonable filesizes as any kind of OMX recording is failing with "get_oformat: ffmpeg_video_codec option value mp4:h264_omx is not supported"
Is there any way to fix this on the central server?
5. Does anyone encounter some kind of blurry picture for objects which are more than 2m away? I had expected a better picture from 8MP to be honest

nighter83, 2018-08-09:
hm no feedback at all on any of the issues :(

alphanumeric007, 2018-08-10:
Running Motion eye on a Pi Zero W I had the streaming issues. Almost unusable to be honest. Lots of lags and really jittery. Motion detection didn't seem to work either, not reliably anyway.
Much better on a Pi 3B (not a +) though. I still had to reduce my Video quality to 50% to get a good stream going that wasn't lagging stuttering etc. Motion detection seems t work now too. 
I'm using the original version 1 Pi camera connected to the CSI connector.

kirillsimin, 2020-01-09:
I am also getting similar issues. Running motionEye on Ubuntu 18.04

Logs:

`[1:ml1:Cars] [ERR] [ENC] get_oformat: ffmpeg_video_codec option value mp4:h264_omx is not supported
[1:ml1:Cars] [ERR] [EVT] event_ffmpeg_newfile: ffopen_open error creating (new) file [/home/kirill/motioneye/media/Camera2/2020-01-08/20-59-29]: Operation now in progress
`

Any help would be appreciated.

jasaw, 2020-01-09:
@kirillsimin h264_omx encoder is only available on Raspberry Pi platform. If you are indeed using a Raspberry Pi, follow the link below on how to enable h264_omx on your ffmpeg.
https://github.com/ccrisan/motioneye/issues/930#issuecomment-522848452

Not sure which motion version you are running, you may need to patch your motion to disable zero_copy, which is the 2nd part of the instructions in the link.

kirillsimin, 2020-01-10:
@jasaw I see. I'm running motionEye on Ubuntu 18.04, on a NUC. My question might belong to motionEye issues, not motionEyeOS, but I'm getting the exact errors as the OP, so I posted here. Do you know how to change encoding for motion?