motioneye-project/motioneyeos #1610: Rebootloop - wired connection

Boelen, 2018-08-06:
Hi there

My motionEyeOs is working for 3 months now.
Today I restarted it (sorry i can't see what version it is - but i thought it was the latest) & i got the next error.
(Error Fetching Interface connection information : device not found ) -- see image.

The weird thing is, you see the connection going up after this message so the connection is established. But its to late so it reboots..

Can i delay the reboot? Or wait till the connection is established?
How can i fix this? Do i need to reinstall motion eye os? That would be shitty :) :)

Image : 
[image]

Greetings

BrutalCoding, 2018-08-08:
I've got the same issue, same log. I cleaned up the storage though by deleting media files, but still getting the other error about wired connection not available. I've setup a static ip in motioneye (also on the router) but still getting this, after removing the content of static_ip.conf I still get this error. I then tried it with a UTP cable to my router (cable works, tested) , still getting the error. 

It seems like something else is happening?

avellanet, 2018-08-13:
First time user and newbie.  I am having the same problem on my Rasp3B+...the ethernet port is not recognized, no lights.  It loaded just fine on my PiZeroW.

ccrisan, 2018-08-14:
@avellanet did you just try the same SD card in a PI3B+ and in a Zero W and you saw that it worked on the latter but did not work on the former?

Boelen, 2018-08-25:
Hi there Crrisan
i reinstalled Motion Eye os 17 days ago because the error didn't go away.
After that i configured everything & it went ok to yesterday.
I needed a reboot from the device so i pressed the 'reboot' button on the web interface.

Now i get the same problem again!
Its not fun to install it over & over & over & over .... Otherwise i need to search to something else.
Also using a raspberry pi by the way.

Greetings
Tom

ccrisan, 2018-08-26:
@Boelen you can always search for something else. I don't insist that you use motionEyeOS. Anyways, back to our main issue:
 
* you ignored our issue template; that's fine, as long as you provide all the information that we're asking for
* the "no space left on device" message that you get usually indicates what @BrutalCoding mentioned -  you should clean up your data partition by removing some media files; once you've done that, if the problem persists, please reattach your logs

Boelen, 2018-08-28:
Hi there @ccrisan 

Brutalcoding is getting the same error.
How can i clean up the data partition by using windows? It is in a reboot loop so i can't delete files with the interface on the raspberry pi.

& Still i think it is weird. I'm not saving any videos or pictures on the raspberry pi. Its just running.

BrutalCoding, 2018-08-28:
Hi @Boelen, 

Just want to give you a heads up that I've simply reinstalled the os and since then had no issues anymore. It probably had something to do with the storage getting too full (it was) but removing all media files didn't help me unfortunately. 

Anyways I suggest you to run a live usb stick with something like Ubuntu, and then you'll be able to access the sd card and read/write with it.

pie314isi, 2018-12-20:
I have a similar problem.

raspberry Pi 3B+
new 16G high speed brand name SDcard
new image of motioneyeos-raspberrypi3-20181209.img 
wired connection. Router is set up to assign IPs over DHCP.

works fine. successfully gets an IP from the router the network, set up cameras, do motion detection, recording etc. I can ssh in and use it all successfully. I love it, by the way.

on first reboot, it goes into a boot cycle with:

Configuring wired network: no link
network: no network connection available, calling panic action
panic: rebooting in xx seconds (caused by network)

I've tried this a number of times, reimaging each time. It goes into the boot loop with even the most minimal change and then clicking "reboot" in the advanced settings.

Anyone got ideas?

ccrisan, 2018-12-26:
@pie314isi do you disconnect your board from the wired network at anytime?

pie314isi, 2018-12-26:
@ccrisan no, it was plugged in to the wired network the whole time.

tok25, 2019-03-15:
hello,
I have the same problem
Raspberry Pi Zero W, Memory card 16go
Motioneyes ver.20190119
[image]
Greetings
JA

ccrisan, 2019-03-15:
@tok25, properly preconfigure your wifi network before booting the os.

tok25, 2019-03-15:
Perfect with [https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration](url)
Thank you very much

closed 2019-03-15 by ccrisan