motioneye-project/motioneyeos #1611: Saving Files to motioneye from (Building From Source) Rpi

rockman55, 2018-08-06:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (0.39.2 motionEyeOS 20180627).

### Board Model

I am using the following board/model: (Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (Raspberry PI).

### Network Connection

My motionEyeOS unit is connected to the network via: (Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 
 * (e.g. External Hard Disk
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/2264424/motioneye.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/2264425/boot.log)

)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
I have created a Pi camera setup using the instructions for Building From Source, then added some files for Pan Tilt capability. I want to be able to save the videos and jpg to the main motioneyeos system. I have tried to save the files using the option in File Storage, (Custom Path, and Network Save does not seem to be available. What if any options do I have to save these files on the primary motioneyeos system and not the host setup that I built from source?

rockman55, 2018-08-21:
Thanks for your reply, I will try to explain my concern. I have 11 separate cameras installed on separate Pi’s that either have Motion or MotionEye installed. I have a primary Pi that monitors all of these cameras and I have a 256GB USB drive install that I want to save all the images and or jpg’s. On the camera in question I have used “Install On Raspbian”. I did this install because I wanted to be able to use Action Buttons to move a Pan Tilt Camera. All my Actions Buttons work great, and I can access them through the main MotionEye system. However, when I try to save the files to the Pan Tilt Camera I go not get an option to save the files on the Main MotionEye system. (I have attached two screen shots of what I hope explains my problem. The PanTilt Cam.jpg is the system that I installed the “Install ON Raspbian” pi the other GuestRoomCam.jpg is using Motion.) Please note in the File Storage section on the GuestRoomCam the option is available for the Storage Device to be SanDisk Cruzer (/dev/sda1), and on the PanTilt Cam that option is not available to store the files on the SanDisk Cruzer, I only can get the Custom Path or the location of the SD card. I have tried to use the Custom Path option and the Root Directory as /data/media/sda1 but this only saves the files to the PanTilt Camera and not my main MotionEye OS system that has the USB drive.

Hopefully this will better explain what I am trying to do.

Thanks,

Danny

 

I don't get this. As far as I understand, you built motionEyeOS from source and then installed the resulting image on a Raspberry PI or a similar board. How could it save anything on the host you used to build the image? What connection do the two still have after you have installed and run the image on your board?

ccrisan, 2018-08-21:
Now I understand what your use case is and what your problem is. There's nothing with the fact that you have built your OS from source. Unfortunately there's no easy way to save media files generated by "slaves" on the master. You could hack the requests between the master and the slaves but that's going to be a some serious work from your side. And my time won't allow me to help you with that.