motioneye-project/motioneyeos #1613: pi 3B+ image?

talormanda, 2018-08-09:
Is there going to be an official image of the OS for the raspberry pi 3B+? I would like to install it on the SD card without having to load it on top of raspbian.

zafuquazar, 2018-08-09:
Hmm, I installed the pi3 image on a 3B+ and it seemed to work fine. I do have connectivity issues, but I believe that is my network rather than the pi

ccrisan, 2018-08-09:
@talormanda the Pi3 image is supposed to work with Pi3B+ as well. I updated the [Supported Devices](https://github.com/ccrisan/motioneyeos/wiki/Supported-Devices) Wiki page.

closed 2018-08-09 by ccrisan

talormanda, 2018-08-09:
@ccrisan Ohhhh okay lol. I did not think it would be unless stated in supported devices. I will certainly use this now. Thanks!