motioneye-project/motioneyeos #1614: motioneyeos problem Add network camera

constantine10, 2018-08-09:
motionEye Version | 0.39.2
-- | --
Motion Version | 4.1.1
OS Version | motionEyeOS 20180627

### motionEyeOS Version

I am running motionEyeOS version: 
motionEye Version 0.39.2
Motion Version 4.1.1
motionEyeOS 20180627

### Board Model

I am using the following board/model:  Raspberry PI 3B.

### Camera

I am using the following type of camera: Network Camera
My camera model is: hikvision network camera with url stream rtsp

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
 External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

[dmesg.log](https://github.com/ccrisan/motioneyeos/files/2275025/dmesg.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/2275026/boot.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/2275027/messages.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/2275028/motioneye.log)

Hello;
my problem is the camera box to add a new network camera is empty
[image]

Thanks for your help

Kywalh, 2018-08-09:
You need to enter a valid URL before the droplist is populated.

closed 2018-08-09 by ccrisan

constantine10, 2018-08-09:
thank you very much
problem solved